SpikeInterface/spikeinterface #751: Issue "spyking-circus: the command is not found."

MNS57, 2022-06-28:
Even if we could edit the "run_spykingcircus.sh" file, this message continue:
     spyking-circus: the command is not found.

How we should do?

If we type in Linux directly ... it runs properly.
     #!/bin/bash
     conda activate circus
     spyking-circus spykingcircus_output/recording.npy -c 12

alejoe91, 2022-06-29:
Hi @MNS57 

Can you please:
1. Explain what you're trying to do
2. Explain how you are running python (E.g. terminal, notebook)
3. Provide your script (copy paste is better)
4. Paste the full error message

In point 2, make sure you are running your script from the `circus` environment!

Thanks, Alessio

MNS57, 2022-07-05:
Dear Alessio,

Actually, I think the topic we previously asked is running well, so it seems to be (probably) fine.

We would like to ask this following different question... 

**How we can extract position of estimated neurons correctly?**

We could get positions of neurons by "st.localize_units" by the following command... 

--------------------------------------------------------------------------------------------------------
unit_locations_com = st.localize_units(we, method="center_of_mass") # we given by "si.extract_waveforms"
x0 = unit_locations_com[:,0]
y0 = unit_locations_com[:,1]
--------------------------------------------------------------------------------------------------------

but...
there are several zero and NaN components, and we wonder how we should eliminate them like....?
If we eliminate zero and NaN components on the second line... there is slight difference ("only 1") with the
number of neurons extracted from "recording.result-merged.hdf5".

Are we better to eliminate "22.456094757523307" too in this example?     Thanks!

	0
0	22.456094757523307   <------
1	22.974765510397535
2	22.39459511533562
3	23.05274408014745
4	22.78789513739714
5	23.29287082658429
6	22.861351036675043
7	22.394401839701928
8	NaN
9	23.381451883134048
10	22.80264190245354
11	23.297553766574794
12	22.32607806988277
13	23.103376126839528
14	NaN
15	NaN
16	22.288422669756677
17	22.33667320123922
...

alejoe91, 2022-07-05:
Hi @MNS57 

That is the correct way to localize units (note that `localize_units` is deprecated now in favor of `compute_unit_locations` - which will be included in the next release).

What is the layout of the probe? It is very strange that you get NaNs in the localization. Can you inspect the template with `plot_unit_templates`?

samuelgarcia, 2022-07-05:
@MNS57 localize units have 2 main methods:
  * "center_of_mass" : super easy and fast. It is the baricenter of amplitudes of the templates over channels.
    This is super strange that you get nan with this methods.
    This method is planar only (2d)
  * "monopolar_triangulation" : it optimise a model of the position in 3D and the impatc peak-to-peak values of some electrodes.

closed 2022-07-19 by alejoe91

MNS57, 2022-07-19:
Thanks for your answer.

By seeing deeply our code, we could find where problem exist.

Let us to ask question in the future if we face any troubles.  Thanks.