SpikeInterface/spikeinterface #752: Not passing some arguments to neo in `BlackrockRecordingExtractor`

h-mayorquin, 2022-06-28:
The `BlackrockRecordingExtractor` only passess `filename` to neo.

https://github.com/SpikeInterface/spikeinterface/blob/153c5fd6c498f52419a5ee1f46808c61436fd1ef/spikeinterface/extractors/neoextractors/blackrock.py#L24-L27

However, there are plenty or arguments such as, `nsx_overide`, that are not propagated:

https://github.com/NeuralEnsemble/python-neo/blob/10c28845a7b74ecbaf53b7a7c93d133540100a7c/neo/rawio/blackrockrawio.py#L128-L129

This used to be propagated in `spikeextractors`:
https://github.com/SpikeInterface/spikeextractors/blob/48a86a048a4e6c5b9d669fd4f3bbf58caa97583a/spikeextractors/extractors/neoextractors/blackrockextractor.py#L34-L38

Is there a reason for this or it was an omission?

samuelgarcia, 2022-06-29:
Hi Ramon,
This `nsx_overide` and `nsx_to_load` is somehow the same use as `stream_id=...`.
It is a bit compicated for the end user.
`nev_override` is not used on SI side.
So I think I did this to simplify user life.

What is the use case you have in mind to propagate this args ?

h-mayorquin, 2022-06-29:
OK, Yes, after loading a file in a folder with multiple files with `ns{x}` extension I realize now how that works. So, the stream is equivalent to the x in the `nx{x}` extension, correct?

Moreover, right now, it is not possible to load `nev` files, correct? #753

alejoe91, 2022-08-02:
Implemented SortingExtractor here https://github.com/SpikeInterface/spikeinterface/pull/856

closed 2022-08-02 by alejoe91