SpikeInterface/spikeinterface #763: get_noise_levels with return_scaled

yger, 2022-06-30:
Just a quick question. Why is it not possible to use the get_noise_levels without sometimes explicitly stating return_scaled=False. What kind of recordings are triggering such an error. Can it be such that by default, the function does not need any argument to work?

samuelgarcia, 2022-07-01:
Salut Pierre.
Some recording do not have scaling because they do not have gain/offset properties.
For instance they are already scaled because store in float32 in that case return_scaled=False is already scaled and so return_scaled=True do not make sens.

yger, 2022-07-01:
Ok, I got the spirit, but why not considering then that the gain is 1? Because this is strange that for the tests of the sorters, one need to specify such a flag to avoid crashes

alejoe91, 2022-07-01:
You can check if a recording has scaled traces: `recording.has_scaled_traces()`

closed 2022-08-02 by alejoe91