SpikeInterface/spikeinterface #768: Error when running klusta sorter in container mode

chyumin, 2022-07-01:
I'm getting this error when trying to run `klusta` sorter in docker or singularity mode

The sorter runs well, this happens when SI tries to extract the container's output

```---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
Input In [4], in <cell line: 1>()
----> 1 sorting = si.run_sorter(
      2     'klusta',
      3     recording,
      4     verbose=True,
      5     delete_output_folder=False,
      6     singularity_image=True
      7 )
      8 sorting

File ~/CatalystNeuro/spikeinterface/spikeinterface/sorters/runsorter.py:132, in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, singularity_image, with_output, **sorter_params)
    130         else:
    131             container_image = singularity_image
--> 132     return run_sorter_container(
    133         container_image=container_image,
    134         mode=mode,
    135         **common_kwargs,
    136     )
    138 return run_sorter_local(**common_kwargs)

File ~/CatalystNeuro/spikeinterface/spikeinterface/sorters/runsorter.py:565, in run_sorter_container(sorter_name, recording, mode, container_image, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, extra_requirements, **sorter_params)
    563 else:
    564     if with_output:
--> 565         sorting = SorterClass.get_result_from_folder(output_folder)
    567 if delete_output_folder:
    568     shutil.rmtree(output_folder)

File ~/CatalystNeuro/spikeinterface/spikeinterface/sorters/basesorter.py:281, in BaseSorter.get_result_from_folder(cls, output_folder)
    277 if bool(log['error']):
    278     raise SpikeSortingError(
    279         "Spike sorting failed. You can inspect the runtime trace in spikeinterface_log.json")
--> 281 sorting = cls._get_result_from_folder(output_folder)
    283 recording = load_extractor(output_folder / 'spikeinterface_recording.json')
    284 if recording is not None:
    285     # can be None when not dumpable

File ~/CatalystNeuro/spikeinterface/spikeinterface/sorters/klusta/klusta.py:155, in KlustaSorter._get_result_from_folder(cls, output_folder)
    153 @classmethod
    154 def _get_result_from_folder(cls, output_folder):
--> 155     sorting = KlustaSortingExtractor(file_or_folder_path=Path(output_folder) / 'recording.kwik')
    156     return sorting

File ~/CatalystNeuro/spikeinterface/spikeinterface/extractors/klustaextractors.py:82, in KlustaSortingExtractor.__init__(self, file_or_folder_path, exclude_cluster_groups)
     79         cs_to_exclude.append(ec.lower())
     81 for channel_group in kf_reader.get('/channel_groups'):
---> 82     chan_cluster_id_arr = kf_reader.get(f'/channel_groups/{channel_group}/spikes/clusters/main')[()]
     83     chan_cluster_times_arr = kf_reader.get(f'/channel_groups/{channel_group}/spikes/time_samples')[()]
     84     chan_cluster_ids = np.unique(chan_cluster_id_arr)  # if clusters were merged in gui,

TypeError: 'NoneType' object is not subscriptable
```

alejoe91, 2022-07-19:
@chyumin does this happen for all recordings? Maybe something just went wrong!

chyumin, 2022-07-24:
@alejoe91 This happenned on toy data when running tests from spikeinterface-dockerfiles, but I also tested to run separately on a notebook and I got the same error.
With real data is working well

alejoe91, 2022-07-25:
@chyumin can you try to install the sorter locally and see if the same error appears? How long is the test recording?