SpikeInterface/spikeinterface #770: File path not found sorting in container

LiesDec, 2022-07-01:
The filepats of my recording is not found when running sorting algorithm using a docker file.
```
sorting_TDC = ss.run_tridesclous(
    recording,
    output_folder="tridesclous_output",
    docker_image="spikeinterface/tridesclous-base:1.6.4",
)
```
The way it tries to find it is by using this method (see [#L170-L207](https://github.com/SpikeInterface/spikeinterface/blob/master/spikeinterface/sorters/runsorter.py#L170-L207))

```
def find_recording_folder(d):
    if "kwargs" in d.keys():
        # handle nested
        kwargs = d["kwargs"]
        nested_extractor_dict = None
        for k, v in kwargs.items():
            if isinstance(v, dict) and is_dict_extractor(v):
                nested_extractor_dict = v
        if nested_extractor_dict is None:
            folder_to_mount = find_recording_folder(kwargs)
        else:
            folder_to_mount = find_recording_folder(nested_extractor_dict)
        return folder_to_mount
    else:
        for k, v in d.items():
            if "path" in k:
                # paths can be str or list of str
                if isinstance(v, str):
                    # one path
                    abs_path = Path(v)
                    if abs_path.is_file():
                        folder_to_mount = abs_path.parent
                    elif abs_path.is_dir():
                        folder_to_mount = abs_path
                elif isinstance(v, list):
                    # list of path
                    relative_paths = []
                    folder_to_mount = None
                    for abs_path in v:
                        abs_path = Path(abs_path)
                        if folder_to_mount is None:
                            folder_to_mount = abs_path.parent
                        else:
                            assert folder_to_mount == abs_path.parent
                else:
                    raise ValueError(
                        f'{k} key for path  must be str or list[str]')
            return folder_to_mount
```
But that gives me the following error:
`UnboundLocalError: local variable 'folder_to_mount' referenced before assignment`

`d` in my case (neuropixel 2.0 dual recording) looks like this:

```
{'class': 'spikeinterface.extractors.neoextractors.openephys.OpenEphysBinaryRecordingExtractor',
 'module': 'spikeinterface',
 'kwargs': {'all_annotations': False,
  'stream_id': '0',
  'folder_path': '/Volumes/FK_Fileserver1/Project_NP_optogenetics/data/raw/NP6/2022-06-29_17-15-43'},
 'dumpable': True,
 'version': '0.94.1.dev0',
```

So I think the line `return folder_to_mount` should be one tab up or outside the `for loop` :)

samuelgarcia, 2022-07-01:
@yger a the same problem. I will have a look

alejoe91, 2022-07-01:
@LiesDec can you paste the full dictionary?

LiesDec, 2022-07-01:
@alejoe91 Of course, here it is :) 

```
{'class': 'spikeinterface.extractors.neoextractors.openephys.OpenEphysBinaryRecordingExtractor',
 'module': 'spikeinterface',
 'kwargs': {'all_annotations': False,
  'stream_id': '0',
  'folder_path': '/Volumes/FK_Fileserver1/Project_NP_optogenetics/data/raw/NP6/2022-06-29_17-15-43'},
 'dumpable': True,
 'version': '0.94.1.dev0',
 'relative_paths': False,
 'annotations': {'is_filtered': False},
 'properties': {'group': array([0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
         0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
         0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
         0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
         0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
         0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
         0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
         0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
         0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
         0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
         0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
         0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
         0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
         0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
         0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
         0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
         0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
         0, 0, 0, 0, 0, 0, 0, 0, 0, 0]),
  'location': array([[   0.,    0.],
         [  32.,   15.],
         [   0.,   30.],
         [  32.,   45.],
         [   0.,   60.],
         [  32.,   75.],
         [   0.,   90.],
         [  32.,  105.],
         [   0.,  120.],
         [  32.,  135.],
         [   0.,  150.],
         [  32.,  165.],
         [   0.,  180.],
         [  32.,  195.],
         [   0.,  210.],
         [  32.,  225.],
         [   0.,  240.],
         [  32.,  255.],
         [   0.,  270.],
         [  32.,  285.],
         [   0.,  300.],
         [  32.,  315.],
         [   0.,  330.],
         [  32.,  345.],
         [   0.,  360.],
         [  32.,  375.],
         [   0.,  390.],
         [  32.,  405.],
         [  32.,  435.],
         [   0.,  465.],
         [  32.,  465.],
         [  32.,  510.],
         [  32.,  600.],
         [   0.,  645.],
         [  32.,  660.],
         [   0.,  705.],
         [  32., 1140.],
         [  32., 1185.],
         [   0., 1245.],
         [  32., 1260.],
[... 420 lines cut ...]
         0.19499999, 0.19499999, 0.19499999, 0.19499999]),
  'offset_to_uV': array([0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
         0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
         0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
         0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
         0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
         0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
         0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
         0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
         0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
         0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
         0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
         0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
         0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
         0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
         0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
         0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
         0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
         0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
         0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
         0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
         0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
         0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
         0., 0., 0., 0., 0., 0., 0., 0., 0., 0.])}}
```

alejoe91, 2022-07-01:
Actually, we'll make a PR! :) give us a few minutes!
I think we never spot this because it only happens when the path is not the first kwarg

LiesDec, 2022-07-01:
@alejoe91 Okey thank you. Yes, I was thinking that indeed. Why do you think the first argument in my case in not path? Does that have something to do with the different streams of data?

alejoe91, 2022-07-01:
I think this was introduced by a recent change in NEO that prepends the `all_annotations` and `stream_id` to any other kwarg:
```
'kwargs': {'all_annotations': False,
  'stream_id': '0',
  'folder_path': '/Volumes/FK_Fileserver1/Project_NP_optogenetics/data/raw/NP6/2022-06-29_17-15-43'},
```

We are working on a more general and elegant fix ;)

alejoe91, 2022-07-01:
@LiesDec can you try this PR? https://github.com/SpikeInterface/spikeinterface/pull/773

This also allows for files to be in multiple folders