SpikeInterface/spikeinterface #789: Recordings in study looks like filtered

yger, 2022-07-06:
I don't think this is normal, but when creating a study, the raw binary files in the raw folder seems to be filtered ("is_filtered": true in the .json file). But is it the case, really ? Because I think the sorters are filtering the files, ignoring this flag. So I guess this is just an error while setting the flag for a newly created binary file

alejoe91, 2022-07-06:
It should depend on the recording!

yger, 2022-07-06:
This is for a MEArec recording, so could it be that they are already filtered? Hum.... I'll check

alejoe91, 2022-07-06:
Yes MEArec recordings are filtered by default! You can check the extractor ;)

yger, 2022-07-06:
Arg... But then, all the sorters are filtering by default, am I right? So recordings are filtered twice when making a study from MEArec recordings?

yger, 2022-07-06:
I was trying to compare circus 2 (based on sorting components) with KS. But then, KS is always filtering the data in a study, while circus can detect (thanks to your nice framework) if recording is filtered or not, and perform accordingly the filtering. This is why I noticed the discrepancy. Not a big deal, but it might affect quite a lot of studies we performed before with the study object, since I was not aware of such a behavior... I always thought MEArec recordings were raw and unfiltered. My bad

yger, 2022-07-06:
Actually, then @alejoe91  there might be a bug in MEArec, because my scripts are generated with the option

recordings_params["recordings"]["filter"] = False

But then, indeed, the flag is_filtered is set to True in the MEArec recordings. There is an error in one of the two softwares. Either they are filtered (and the line is not properly handled), either they are not, but SI thinks they are...

yger, 2022-07-06:
Well then, but what is annoying is that SI consider files from MEArec as filtered, even if this is not the case. If one day MEArec adds LFP content, then this is problematic. Moreover, I can certify you (I just tested it) that filtering or not the file change quite a lot the peak detection. So one must know what is done, and be sure all sorters are doing the same

closed 2022-08-01 by alejoe91