SpikeInterface/spikeinterface #803: Allow user to specify temporary directory for sorting

lfrank, 2022-07-08:
MountainSort, and perhaps other sorters, use a specific temporary directory to write out data. Right now there is no way to specify that temporary directory as a parameter to the sorter. Run times for very large datasets are greatly accelerated if we can specify a local SSD / NVME drive as the temporary directory, and it would be very helpful if that parameter could be added and passed to the sorters when they are called.

magland, 2022-07-08:
@alejoe91 @lfrank 
I just published new mountainsort4 (1.0.2)
There is now a tempdir parameter for mountainsort4 function
See
https://github.com/magland/mountainsort4/blob/3d9b6f5d62bb35e117c9cc71eda3e0a8ae2fd39e/mountainsort4/mountainsort4.py#L12-L13

alejoe91, 2022-08-01:
Fixed

closed 2022-08-01 by alejoe91