SpikeInterface/spikeinterface #814: mismatch between phy and si unit ids

AlanoudAlsubhi, 2022-07-12:
Hi, 

I saw the fix for the mismatch between phy and si unit ids. I tried the solution but the problem still exists in my end. 
Any suggestion of what I should try? 

Thanks, 
Alanoud

alejoe91, 2022-07-12:
Hi @AlanoudAlsubhi 

Phy internally uses sequential IDs from 0 to N-1. Since spikeinterface IDs could be different, we save an additional property called `si_unit_id` that should show as an additional column in the phy unit list. 

These `si_unit_ids` are then loaded back in spikeinterface if you load a curated phy output.

alejoe91, 2022-07-19:
@AlanoudAlsubhi is it clearer now?

AlanoudAlsubhi, 2022-08-09:
Yeah but now I'm running into another thing that is confusing me. It might be related to this, not sure. 

I'm using the `spikeinterface.postprocessing.get_template_extremum_channel(waveform_extractor, peak_sign: str = 'neg', mode: str = 'extremum', outputs: str = 'id')`

Some of the extremum channels are not consistent pre and post-curation. In this case it is just the first one.
Pre curation it is (0:0), post curation it is (0:11).  Any ideas? 

[image]

[image]

Thanks

alejoe91, 2022-08-09:
Did you do some merges or just removed noisy units?

AlanoudAlsubhi, 2022-08-09:
I just removed noisy units

alejoe91, 2022-08-09:
Ok, it seems that the mismatch is only for unit `0`, right? 

I think this is happening: 
Unit 0 probably have waveforms which are almost as large on two channels (0 and 11). When you extract waveforms, you subsample them (by default 500), so it is possible that different subsampling make the peak channel fall in one of the two channels. 

Can you try to re-extract waveforms adding the `max_spikes_per_unit=None`? This forces the waveform extraction on all waveforms, so the extremum channels should correspond. 

Let me know! 

PS: you can also try to plot the templates!

AlanoudAlsubhi, 2022-08-09:
Thanks for the suggestions! Unfortunately, nothing changed. 
But there is something interesting from plotting the templates Pre and Post. There seems to be something wrong with unit 0 post curation, `sorting_SC_phy_curated = spikeinterface.extractors.read_phy('phy_SC/', exclude_cluster_groups=['noise'])`. 
[image]
[image]

alejoe91, 2022-08-09:
Yes indeed it looks wrong..can you try to compare the spike trains? If you share the phy folder I'll take a look

AlanoudAlsubhi, 2022-08-09:
[https://drive.google.com/drive/folders/1gQgiNWabO1npZYje4LfG7wkUrTG33K8T?usp=sharing](url)

alejoe91, 2022-08-10:
@AlanoudAlsubhi thanks for the data and thanks for spotting this! Indeed there was a bug!

It is fixed in this PR: https://github.com/SpikeInterface/spikeinterface/pull/876

Can you try to install from there?

AlanoudAlsubhi, 2022-08-11:
This actually fixed the issue. 
Thank you!

closed 2022-08-12 by alejoe91