SpikeInterface/spikeinterface #815: Difference between old (0.13.0) and new (0.94.0) bandpass filter

max-c-lim, 2022-07-12:
Can someone please elaborate the difference between the old API's (version 0.13.0) and the new API's (version 0.94.0) bandpass filter and which is better?

When I use the old API's bandpass filter `spiketoolkit.preprocessing.bandpass_filter` with the default parameters, I get different results on the same recording from the new API's bandpass filter `spikeinterface.toolkit.preprocessing.bandpass_filter`. 

My recording has scaled traces, so with the new API, I use `spikeinterface.toolkit.preprocessing.scale` to scale my recording and convert to float32 before the bandpass filter. The old API appears to do this automatically.

Thanks!

alejoe91, 2022-07-12:
Hi @max-c-lim 

The new API uses scipy as default, while the old one used a custom FFT based filter. The implementation of the new API is more stable and efficient. 

Regarding scaling, if you loaded channel gains then each recording can return either the raw or the scaled traces, so you don't need to add a scale processing. You simply add the `return_scaled=True` when calling the `get_traces`.

max-c-lim, 2022-07-12:
Thanks for the clarification!

I noticed that if I do not manually scale the recording and convert from uint16 to float32 before applying the bandpass filter, `get_traces` applies the bandpass filter on the uint16 data, scales the data, and then converts to float32. When the bandpass filter is applied on the uint16 data, it outputs many zeros, and important data is lost. I tried setting `dtype="float32` when calling the bandpass filter, but this does not convert the recording from uint16 to float32 before the bandpass filter, only after data is already lost.

alejoe91, 2022-07-12:
So this is a specific issue fo uint32 that we have fixed recently: https://github.com/SpikeInterface/spikeinterface/pull/798

Can you try to install from master?

max-c-lim, 2022-07-12:
Yes, the problem is fixed! Thank you!

For Kilosort2, do you recommend scaled or unscaled traces for the recording.dat file? I have been manually scaling the data so that Kilosort2 analyzes scaled data.

alejoe91, 2022-07-12:
I recommend unscaled, since it doesn't make a difference for spike sorting and the temporary data is smaller. Note that when you extract waveforms after spike sorting the traces are scaled automatically, because then you want to look at uV ;)

max-c-lim, 2022-07-12:
The recording.dat file is in int16. The traces of my unscaled recordings are fairly small, and almost all of the data is rounded to 0 when converting from float32 to int16. Will this cause Kilosort2 to perform worse?

samuelgarcia, 2022-07-13:
@max-c-lim : also other diff between 0.13 and new APi on filtering:
  * the new API use sosfilter
  * the 0.13 api were not handling margin super properly.

max-c-lim, 2022-07-13:
@alejoe91 
I use MaxWell Biosystems to make recordings. They are stored as .raw.h5 files and loaded with `MaxwellRecordingExtractor`. The raw recordings are `uint16`. The gain is `6.295`, and the offset is `0`. 

Yes, when I cast to `int16` most of the data are zeros because the traces are small values that are rounded to 0. This does not happen when I use `spikeinterface.toolkit.preprocessing.scale` to scale the recordings before the cast since the traces are larger values. I believe that this is a quirk of my specific recordings rather than a problem with SpikeInterface. 

[image]

I manually use `BinaryRecordingExtractor.write_recording` to convert each recording to a .dat file with type `int16` for input to Kilosort2. I looked at the 0.94.0 source code, and I believe that this function calls `get_traces` without allowing the user to specify `return_scaled`. Thus, I use `spikeinterface.toolkit.preprocessing.scale` to scale the recording before converting it to force the `get_traces` call to return scaled traces.

It would be nice to have `return_scaled` as a parameter for `BinaryRecordingExtractor.write_recording`.

Thanks for all your help!

alejoe91, 2022-07-19:
Casting to int16 should not produce zeros. The `uint16` values CANNOT be close to zero, otherwhise they would be already zero since `unint` is an integer type. Does it make sense?

The problem of casting to int is only about overflow errors, but you could fix that by casting to `int32` instead.

max-c-lim, 2022-07-21:
Yes, this makes much more sense. Thanks for helping me so much!

closed 2022-07-21 by max-c-lim