SpikeInterface/spikeinterface #818: ProbeInterface: Defining a probe manually and attaching it to recording

TdSNielsen, 2022-07-15:
Hi,

I am probably making some basic mistake, but I have tried to follow the ProbeInterface documentation as best I could, in order to add the probe information required by the sorters. I am using custom made probes (from MicroProbes) and therefore am defining them manually. In my test data, I am using a 16-channel MEA consisting of 16 needles with the ends de-insulated, arrayed in two colums of 8 electrodes with 1 mm between each row and column (and the active sites inserted 2 mm below the surface of the cortex). The sorters therefore just need to know that all channels are independent from each other.

I have tried the following to set up the probe, based on the ProbeInterface documentation I could find:
-------------------------------------------------------
from probeinterface import Probe, ProbeGroup
positions = np.zeros((16, 2))
positions[:, 0] = [0, 1000, 2000, 3000, 4000, 5000, 6000, 7000,
                   0, 1000, 2000, 3000, 4000, 5000, 6000, 7000]  # row position [um]
positions[:, 1] = [0, 0, 0, 0, 0, 0, 0, 0,
                   1000, 1000, 1000, 1000, 1000, 1000, 1000, 1000]  # column position [um]
probe = Probe(ndim=2, si_units='um')
probe.set_contacts(positions=positions,
                   shapes='circle',
                   shape_params={'radius': 50},
                   shank_ids=[0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15])
probe.set_contact_ids([0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15])
probe.set_device_channel_indices([0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15])
-------------------------------------------------------
To add this information to my recording, I call:
-------------------------------------------------------
recording = recording.set_probe(probe)
-------------------------------------------------------
However, when I call recording.get_probe() or ss.run_sorters, I get an error that probe information is missing and I also don't see this information in the recording object when running the debugger...

I hope you can point me to what I'm doing wrong :)

Cheers,
Thomas

alejoe91, 2022-07-15:
Hi Thomas,

The probe definition looks good. Since you want to sort single channels separately, note that you have to use `by_shank=True` when setting the probe:
```
recording = recording.set_probe(probe, by_shank=True)
print(recording.get_channel_group())
# should print the different groups (shanks)
```

Can you try to print `recording.get_probe()` after setting it? 
What's the exact error you are getting?

Finally, to spike sort by group, you should use the [`run_sorter_by_property` ](https://github.com/SpikeInterface/spikeinterface/blob/master/spikeinterface/sorters/launcher.py#L46)function:
```
sort_by_group = ss.run_sorter_by_property(sorter_name="tridesclous", recording=recording, grouping_property="group", working_folder="some-folder")
```

TdSNielsen, 2022-07-15:
thanks for the quick answer, Alessio!

When I try to add the _by_shanks_ command, I get an error that it doesn't exist:
_TypeError: BaseRecording.set_probe() got an unexpected keyword argument 'by_shank'_

No, before trying to add _by_shank_, calling _print(recording.get_probe())_ caused an error (no probe information)...

Thanks, I will try to look into _run_sorter_by_property_ :)

alejoe91, 2022-07-15:
oups my bad:

It should be: `recording = recording.set_probe(probe, group_mode="by_shank")`

It's very strange that it gives the no probe information error..I'll quickly try to reproduce the issue on my side.

alejoe91, 2022-07-15:
So your code seems to be working on my side:

[image]

Note that I'm using `rec.set_channel_groups()` directly instead of loading the shank ids

TdSNielsen, 2022-07-18:
Thanks for the support, Alessio!

_recording.get_probe()_ is also working on my side now, and _ss.run_sorter_by_property_ works for tridesclous, though it exited with a warning: _UserWarning: Cannot propagate registered recording to UnitsAggregationSorting_

One question: If I run each sorter with _ss.run_sorter_by_property_, will I be able to later load all the sorting data from the sorting folders and combine it in postprocessing (since I can't use list input as with _ss.run_sorters_)? My intention was to copy the approach you used in the SpikeInterface paper, where you used multiple sorters to improve accuracy... 

Cheers,
Thomas

alejoe91, 2022-07-19:
The `ss.run_sorter_by_property` returns already a Sorting object, but with an additional `group` property for each unit. Since internally the sorting output is distributed across several folders, I recommend saving the aggregated output in a single folder:
```
sort_by_group = ss.run_sorter_by_property(sorter_name="tridesclous", recording=recording, grouping_property="group", working_folder="some-folder")

sort_by_group = sort_by_group.save(fodler="some-other-folder")

# reload later
sort_by_group = si.load_extractor("some-other-folder")
```

closed 2022-10-13 by alejoe91