SpikeInterface/spikeinterface #822: Unable to run the demo python source code in the gettin start tutorial

YangWenfei-philip, 2022-07-15:
When i try to run the code, exception occurred. I tried to resolve the issue by downgrading the numpy version to 1.21.5, but the problem persisted.

Exception has occurred: ValueError
numpy.ndarray size changed, may indicate binary incompatibility. Expected 96 from C header, got 88 from PyObject
  File "/Users/wenfeiyang/Downloads/plot_getting_started.py", line 129, in <module>
    print('Installed sorters', ss.installed_sorters())

alejoe91, 2022-07-19:
Hi @YangWenfei-philip 

Can you try to `pip uninstall herdingspikes` and then reinstall it: `pip install herdingspikes`?

YangWenfei-philip, 2022-08-12:
I managed to resolved this issue, but when I run the data I encountered the following issue:

UserWarning: resource_tracker: There appear to be 9 leaked semaphore objects to clean up at shutdown

My computer has 16 Gb of RAM, and I managed to free up 12 Gb for this task and this still appear. Is there any solution?

alejoe91, 2022-08-12:
Hi,

This is a warning, so don't worry about it ;) 

Can you tell us when this happens? What is the code that is triggering it?

Cheers
Alessio

closed 2022-12-28 by alejoe91