SpikeInterface/spikeinterface #823: Sorting error when using joblib

mfvd, 2022-07-18:
Hello, 

I'm trying to run muntsinsort4 in parallel on a new computer (ubuntu 20.04.4 LTS) and get the error at the bootom if this issue. I've reinstalled joblib but it did not correct the issue. I'm trying to sort a 32 tetrode dataset to which I added probe information as follows:

```python
# define tetrode geometry and channel indices
tts = pi.ProbeGroup()
for i in range(32):
    tetrode = pi.generate_tetrode()
    tetrode.move([i * 50, 0])
    tts.add_probe(tetrode)
# set indices
tts.set_global_device_channel_indices(channel_indices)

# set probe to recording object
rec_tts = rec.set_probegroup(tts, 
                             group_mode='by_probe')
```

For reference the code I'm using for sorting is the following:

```python 
# get dafault params for ms4
default_ms4_params = ss.get_default_params('mountainsort4')
# set params
default_ms4_params['detect_threshold'] = 4
default_ms4_params['detect_sign'] = -1
default_ms4_params['filter'] = False
default_ms4_params['whiten'] = True

# run sorter
sorted_ms4 = ss.run_sorter_by_property(sorter_name = 'mountainsort4',
                                       recording = rec_cached,
                                       working_folder = 'ms4_output',  # 'sort_'+str(session)
                                       grouping_property = 'group',
                                       engine="joblib",
                                       engine_kwargs={"n_jobs": 16},
                                       verbose = False,
                                       **default_ms4_params)

```
_____________________________________________________________________________________

ERROR:

```python
Mountainsort4 use the OLD spikeextractors mapped with NewToOldRecording
Mountainsort4 use the OLD spikeextractors mapped with NewToOldRecording

Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/neuroimaging/anaconda2/envs/si_openephys/lib/python3.9/multiprocessing/spawn.py", line 116, in spawn_main
    exitcode = _main(fd, parent_sentinel)
  File "/home/neuroimaging/anaconda2/envs/si_openephys/lib/python3.9/multiprocessing/spawn.py", line 125, in _main
    prepare(preparation_data)
  File "/home/neuroimaging/anaconda2/envs/si_openephys/lib/python3.9/multiprocessing/spawn.py", line 231, in prepare
    set_start_method(data['start_method'], force=True)
  File "/home/neuroimaging/anaconda2/envs/si_openephys/lib/python3.9/multiprocessing/context.py", line 247, in set_start_method
    self._actual_context = self.get_context(method)
  File "/home/neuroimaging/anaconda2/envs/si_openephys/lib/python3.9/multiprocessing/context.py", line 239, in get_context
    return super().get_context(method)
  File "/home/neuroimaging/anaconda2/envs/si_openephys/lib/python3.9/multiprocessing/context.py", line 193, in get_context
    raise ValueError('cannot find context for %r' % method) from None
ValueError: cannot find context for 'loky'
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/neuroimaging/anaconda2/envs/si_openephys/lib/python3.9/multiprocessing/spawn.py", line 116, in spawn_main
    exitcode = _main(fd, parent_sentinel)
  File "/home/neuroimaging/anaconda2/envs/si_openephys/lib/python3.9/multiprocessing/spawn.py", line 125, in _main
    prepare(preparation_data)
[... 300 lines cut ...]
    539 """Wrapper for Future.result to implement the same behaviour as
    540 AsyncResults.get from multiprocessing."""
    541 try:
--> 542     return future.result(timeout=timeout)
    543 except CfTimeoutError as e:
    544     raise TimeoutError from e

File ~/anaconda2/envs/si_openephys/lib/python3.9/concurrent/futures/_base.py:446, in Future.result(self, timeout)
    444     raise CancelledError()
    445 elif self._state == FINISHED:
--> 446     return self.__get_result()
    447 else:
    448     raise TimeoutError()

File ~/anaconda2/envs/si_openephys/lib/python3.9/concurrent/futures/_base.py:391, in Future.__get_result(self)
    389 if self._exception:
    390     try:
--> 391         raise self._exception
    392     finally:
    393         # Break a reference cycle with the exception in self._exception
    394         self = None

SpikeSortingError: Spike sorting failed. You can inspect the runtime trace in spikeinterface_log.json

```

alejoe91, 2022-07-18:
Can you try `pip install loky`?

mfvd, 2022-07-18:
Hi Allesio,

loky is already installed with version 3.1.0

samuelgarcia, 2022-07-19:
Oups.
Can you try with "n_jobs": 1 and check if it works without multi processing ?

mfvd, 2022-07-19:
@samuelgarcia In the meanwhile I managed to run the sorter by not specifying the ```engine```, neither the ```n_jobs```. but not when multi processing.

Also, caching a recording object, extracting waveforms and exporting to phy are extremely slow and can only be done when not using parallel processing. In fact, when extracting waveforms or exporting to phy, the kernel of my notebook crashes almost immediately.

samuelgarcia, 2022-07-19:
extacting waveforms is another story because for this we do not use joblib.

Can you try run sorter with joblib but with another sorter ?

And can you give us the error when extracting waveforms witj n_jobs>1 ?

mfvd, 2022-07-22:
@samuelgarcia Sorry, I finally managed to do this and get the error below when trying to sort with tridesclous. I included warngins about the recording not having a probe because I defined my tetrodes beforehand (see original post on this thread).

```python
home/neuroimaging/anaconda2/envs/si_openephys/lib/python3.9/site-packages/spikeinterface/core/baserecording.py:466: UserWarning: There is no Probe attached to this recording. Creating a dummy one with contact positions
  warn('There is no Probe attached to this recording. Creating a dummy one with contact positions')
/home/neuroimaging/anaconda2/envs/si_openephys/lib/python3.9/site-packages/spikeinterface/core/baserecording.py:466: UserWarning: There is no Probe attached to this recording. Creating a dummy one with contact positions
  warn('There is no Probe attached to this recording. Creating a dummy one with contact positions')
/home/neuroimaging/anaconda2/envs/si_openephys/lib/python3.9/site-packages/spikeinterface/core/baserecording.py:466: UserWarning: There is no Probe attached to this recording. Creating a dummy one with contact positions
  warn('There is no Probe attached to this recording. Creating a dummy one with contact positions')
/home/neuroimaging/anaconda2/envs/si_openephys/lib/python3.9/site-packages/spikeinterface/core/baserecording.py:466: UserWarning: There is no Probe attached to this recording. Creating a dummy one with contact positions
  warn('There is no Probe attached to this recording. Creating a dummy one with contact positions')
/home/neuroimaging/anaconda2/envs/si_openephys/lib/python3.9/site-packages/spikeinterface/core/baserecording.py:466: UserWarning: There is no Probe attached to this recording. Creating a dummy one with contact positions
  warn('There is no Probe attached to this recording. Creating a dummy one with contact positions')
/home/neuroimaging/anaconda2/envs/si_openephys/lib/python3.9/site-packages/spikeinterface/core/baserecording.py:466: UserWarning: There is no Probe attached to this recording. Creating a dummy one with contact positions
  warn('There is no Probe attached to this recording. Creating a dummy one with contact positions')
/home/neuroimaging/anaconda2/envs/si_openephys/lib/python3.9/site-packages/spikeinterface/core/baserecording.py:466: UserWarning: There is no Probe attached to this recording. Creating a dummy one with contact positions
  warn('There is no Probe attached to this recording. Creating a dummy one with contact positions')
/home/neuroimaging/anaconda2/envs/si_openephys/lib/python3.9/site-packages/spikeinterface/core/baserecording.py:466: UserWarning: There is no Probe attached to this recording. Creating a dummy one with contact positions
  warn('There is no Probe attached to this recording. Creating a dummy one with contact positions')
/home/neuroimaging/anaconda2/envs/si_openephys/lib/python3.9/site-packages/spikeinterface/core/baserecording.py:466: UserWarning: There is no Probe attached to this recording. Creating a dummy one with contact positions
  warn('There is no Probe attached to this recording. Creating a dummy one with contact positions')
/home/neuroimaging/anaconda2/envs/si_openephys/lib/python3.9/site-packages/spikeinterface/core/baserecording.py:466: UserWarning: There is no Probe attached to this recording. Creating a dummy one with contact positions
  warn('There is no Probe attached to this recording. Creating a dummy one with contact positions')
/home/neuroimaging/anaconda2/envs/si_openephys/lib/python3.9/site-packages/spikeinterface/core/baserecording.py:466: UserWarning: There is no Probe attached to this recording. Creating a dummy one with contact positions
  warn('There is no Probe attached to this recording. Creating a dummy one with contact positions')
/home/neuroimaging/anaconda2/envs/si_openephys/lib/python3.9/site-packages/spikeinterface/core/baserecording.py:466: UserWarning: There is no Probe attached to this recording. Creating a dummy one with contact positions
  warn('There is no Probe attached to this recording. Creating a dummy one with contact positions')
/home/neuroimaging/anaconda2/envs/si_openephys/lib/python3.9/site-packages/spikeinterface/core/baserecording.py:466: UserWarning: There is no Probe attached to this recording. Creating a dummy one with contact positions
  warn('There is no Probe attached to this recording. Creating a dummy one with contact positions')
/home/neuroimaging/anaconda2/envs/si_openephys/lib/python3.9/site-packages/spikeinterface/core/baserecording.py:466: UserWarning: There is no Probe attached to this recording. Creating a dummy one with contact positions
  warn('There is no Probe attached to this recording. Creating a dummy one with contact positions')
/home/neuroimaging/anaconda2/envs/si_openephys/lib/python3.9/site-packages/spikeinterface/core/baserecording.py:466: UserWarning: There is no Probe attached to this recording. Creating a dummy one with contact positions
  warn('There is no Probe attached to this recording. Creating a dummy one with contact positions')
/home/neuroimaging/anaconda2/envs/si_openephys/lib/python3.9/site-packages/spikeinterface/core/baserecording.py:466: UserWarning: There is no Probe attached to this recording. Creating a dummy one with contact positions
  warn('There is no Probe attached to this recording. Creating a dummy one with contact positions')
[... cut ...]
    441 self._condition.wait(timeout)
    443 if self._state in [CANCELLED, CANCELLED_AND_NOTIFIED]:

File ~/anaconda2/envs/si_openephys/lib/python3.9/concurrent/futures/_base.py:391, in Future.__get_result(self)
    389 if self._exception:
    390     try:
--> 391         raise self._exception
    392     finally:
    393         # Break a reference cycle with the exception in self._exception
    394         self = None

SpikeSortingError: Spike sorting failed. You can inspect the runtime trace in spikeinterface_log.json

```

Zoe0793, 2022-08-19:
Hello, 
I am experiencing the same "UserWarning: There is no Probe attached to this recording. Creating a dummy one with contact positions" when running:

sorting_TDC = si.run_sorter_by_property("tridesclous", recording_tetrodes, grouping_property="group",
                                        working_folder="tds_by_group", engine="joblib",
                                            engine_kwargs={"n_jobs": 8},**tdc_sorter_params)

This is the case for tridesclous and spyking circus and I do not get the error if the default engine settings are used.

alejoe91, 2023-06-12:
@mfvd @Zoe0793 can we close this?

Zoe0793, 2023-06-12:
I don't have this issue anymore,
Thanks,
Zoe

closed 2023-06-12 by alejoe91