SpikeInterface/spikeinterface #824: exctract amplitudes from phy to spikeinterface

shirquinn, 2022-07-18:
Hi
I'm using kilosort for spike sorting, and you used to have an option of Filter = False that i don't see exist anymore

I tried uploading the phy data to spikeinterface (after doing the spike sorting in matlab directly on the un processed recording) with:
sorting_KS = si.read_phy(sorting_folder, exclude_cluster_groups=['noise'])
i extracted the waveforms with:
we = si.extract_waveforms(recording_loaded, sorting_KS, folder="./sorted/wf_KS2", 
                              max_spikes_per_unit=None, progress_bar=True,load_if_exists=True)
and when i get to extracting amplitudes:
amplitudes = si.compute_spike_amplitudes(we, outputs="by_unit", n_jobs=20, progress_bar=True, 
                                                                                  chunk_duration="1s")

i get this error:
`extract amplitudes: 100%|███████████████████████████████████████████████████████████| 300/300 [00:02<00:00, 124.28it/s]
---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
Input In [119], in <cell line: 1>()
----> 1 amplitudes = si.compute_spike_amplitudes(we, outputs="by_unit", n_jobs=20, progress_bar=True, 
      2                                                                                   chunk_duration="1s")

File ~\AppData\Roaming\Python\Python39\site-packages\spikeinterface-0.94.1.dev0-py3.9.egg\spikeinterface\toolkit\postprocessing\spike_amplitudes.py:181, in compute_spike_amplitudes(waveform_extractor, load_if_exists, peak_sign, return_scaled, outputs, **job_kwargs)
    178     sac.set_params(peak_sign=peak_sign, return_scaled=return_scaled)
    179     sac.compute_amplitudes(**job_kwargs)
--> 181 amps = sac.get_amplitudes(outputs=outputs)
    182 return amps

File ~\AppData\Roaming\Python\Python39\site-packages\spikeinterface-0.94.1.dev0-py3.9.egg\spikeinterface\toolkit\postprocessing\spike_amplitudes.py:156, in SpikeAmplitudesCalculator.get_amplitudes(self, outputs)
    154         spike_times, spike_labels = self._all_spikes[segment_index]
    155         mask = spike_labels == unit_index
--> 156         amps = self._amplitudes[segment_index][mask]
    157         amplitudes_by_unit[segment_index][unit_id] = amps
    158 return amplitudes_by_unit

IndexError: boolean index did not match indexed array along dimension 0; dimension is 3406 but corresponding boolean dimension is 31337`

Is there a way to directly upload the amplitudes from phy to spikeinterface for the quality metrics?
or, is there a way to run kilosort via spikeinterface without using the filter?

Thank you

alejoe91, 2022-07-19:
Hi @shirquinn 

Could you share the zipped phy folder (without the .dat file)? 

We'll take a look. There is no way to upload amplitudes directly, so we need to understand why it is failing.

Cheers
Alessio

shirquinn, 2022-07-19:
Hi @alejoe91 
Thanks for responding so quickly!
Here is the phy folder:
[https://drive.google.com/drive/folders/19I121G0RURpvQerJmAuHMT4AW_05E28d?usp=sharing](url)

alejoe91, 2022-07-19:
@shirquinn actually could you add the dat file as well? Sorry just realized I need it :)

Also I need to know the number of channels, sampling frequency, and data type (I guess int16) (basically, how did you load the `recording_loaded`?)

shirquinn, 2022-07-20:
@alejoe91 here is the link to the phy folder with the dat file
[https://drive.google.com/file/d/1OKN4_DX_oIbfyUaxsPUGe-81_LZl2u2O/view?usp=sharing](url)

the original recording is a binary file recorded with intan amplifier and open ephys acquisition box (int16)
what i am looking to do is spike sort an un-processed data, and also use spikeinterface for extracting waveforms, amplitudes,pc etc. for quality metrics

shirquinn, 2022-07-25:
Hi @alejoe91 
i followed the videos and again the problem remains, this error also doesn't alow me to open the spikeinterface gui
IndexError: boolean index did not match indexed array along dimension 0; dimension is 43898 but corresponding boolean dimension is 298862
This error happens no matter if i sort the recording directly in matlab or through spikeinterface
Only if i sort a filtered recording - this error doesn't happen
i read the recording with :

`recording = si.read_openephys(recording_file)
`
this is a link to the data and phy folder:
[https://drive.google.com/drive/folders/1qsmuSNvlME1-zQyjCySoqXezbo-eA3t3?usp=sharing](url)

I'd really appreciate your input on this
Thank you Alessio

alejoe91, 2022-08-02:
@shirquinn unfortunately the zip file containing the recording seems broken... can you double check on your side?

shirquinn, 2022-08-04:
@alejoe91 can you try this link?

[https://drive.google.com/drive/folders/1qsmuSNvlME1-zQyjCySoqXezbo-eA3t3?usp=sharing](url)

Thanks

alejoe91, 2022-08-04:
> @alejoe91 can you try this link?
> 
> https://drive.google.com/drive/folders/1qsmuSNvlME1-zQyjCySoqXezbo-eA3t3?usp=sharing
> 
> Thanks

No luck unfortunately. The decompression of the zip files fails on my side:
[image]

Maybe just upload the entire folder to drive without zipping it?

shirquinn, 2022-08-04:
sorry about that... i've uploaded the folders without zipping now
https://drive.google.com/drive/folders/1qsmuSNvlME1-zQyjCySoqXezbo-eA3t3?usp=sharing

shirquinn, 2022-08-04:
Hi @alejoe91 , i've uploaded the folders to https://drive.google.com/drive/folders/1qsmuSNvlME1-zQyjCySoqXezbo-eA3t3?usp=sharing

This error happens with all of my recordings (when sorting the unfiltered recording, the amplitude extraction has an error, and the l ratio is not calculated)

Thank you for your help

alejoe91, 2022-08-04:
@shirquinn I'm missing the `spike_times.npy` in the phy folder

alejoe91, 2022-08-04:
But then do you extract the waveforms and amplitudes on the unfiltered traces? or for waveform extraction you filter them? Can you send the full script that you use?

shirquinn, 2022-08-04:
the folder with phy results is called 'sorted', it contains the spiketimes.npy:
https://drive.google.com/drive/folders/18uVOLmSa3uduMgOSP7TJ70NHKDfa4O38?usp=sharing
https://drive.google.com/file/d/15oGY--HDwrQXhDGTZjfnqgiSnE93HGM7/view?usp=sharing

the script:
https://drive.google.com/file/d/1fqG6W61gmeQVVV4hGbruGKgHY41ydl0C/view?usp=sharing

alejoe91, 2022-08-04:
@shirquinn with the latest data you shared I could not reproduce the issue.
[image]

everything seems fine on my side. What version of spikeinterface are you using? Can you try to use the "master" version? We are going to make a release at the end of the month.

shirquinn, 2022-08-04:
@alejoe91 can you try extracting l ratio from these and tell me if its working ok on your end?
i am using the master version... i'll try to reinstall everything

alejoe91, 2022-08-04:
But amplitude extraction didn't fail then, correct?

alejoe91, 2022-08-04:
strange, I used exactly the same code (you can see my screenshot)! Please send me the full script

shirquinn, 2022-08-04:
Sorry @alejoe91 i'm an idiot, i figured out what was the problem, i'll spare us my embarrassment and close this issue :D

closed 2022-08-04 by shirquinn

alejoe91, 2022-08-04:
> Sorry @alejoe91 i'm an idiot, you are the best. i figured out what was the problem, i'll spare us my embarrassment and close this issue :D

ehehehe! no worries! I'm happy to help and don't let yourself down ;)