SpikeInterface/spikeinterface #833: Error when checking installed sorters

mfvd, 2022-07-20:
I'm getting this error when checking the installed sorters with ```ss.installed_sorters()```

```python
ValueError                                Traceback (most recent call last)
Input In [2], in <cell line: 1>()
----> 1 ss.installed_sorters()

File ~/anaconda2/envs/si_openephys/lib/python3.9/site-packages/spikeinterface/sorters/sorterlist.py:48, in installed_sorters()
     45 def installed_sorters():
     46     """Lists installed sorters."""
---> 48     return sorted([s.sorter_name for s in sorter_full_list if s.is_installed()])

File ~/anaconda2/envs/si_openephys/lib/python3.9/site-packages/spikeinterface/sorters/sorterlist.py:48, in <listcomp>(.0)
     45 def installed_sorters():
     46     """Lists installed sorters."""
---> 48     return sorted([s.sorter_name for s in sorter_full_list if s.is_installed()])

File ~/anaconda2/envs/si_openephys/lib/python3.9/site-packages/spikeinterface/sorters/herdingspikes/herdingspikes.py:128, in HerdingspikesSorter.is_installed(cls)
    125 @classmethod
    126 def is_installed(cls):
    127     try:
--> 128         import herdingspikes as hs
    129         HAVE_HS = True
    130     except ImportError:

File ~/anaconda2/envs/si_openephys/lib/python3.9/site-packages/herdingspikes/__init__.py:4, in <module>
      1 from herdingspikes import probe
      3 # from herdingspikes.parameter_optimisation import OptimiseParameters
----> 4 from herdingspikes.hs2 import HSDetection, HSClustering
      6 from herdingspikes.version import __version__, __commit__, base_version
      8 __all__ = ["probe", "OptimiseParameters", "HSDetection", "HSClustering", "__version__"]

File ~/anaconda2/envs/si_openephys/lib/python3.9/site-packages/herdingspikes/hs2.py:7, in <module>
      5 import h5py
      6 import os
----> 7 from .detection_localisation.detect import detectData
      8 from matplotlib import pyplot as plt
      9 from .clustering.mean_shift_ import MeanShift

File ~/anaconda2/envs/si_openephys/lib/python3.9/site-packages/herdingspikes/detection_localisation/detect.pyx:1, in init herdingspikes.detection_localisation.detect()

ValueError: numpy.ndarray size changed, may indicate binary incompatibility. Expected 96 from C header, got 88 from PyObject
```

alejoe91, 2022-07-20:
Uninstall and reinstall `herdingspikes` ;)

mfvd, 2022-07-20:
Thanks! Will do.

mfvd, 2022-07-22:
@alejoe91 just uninstalled and reinstalled herdingspikes and get the same error

alejoe91, 2022-07-22:
Since this is related to HS, maybe you can open an issue on the hs2 repo!

closed 2022-07-25 by mfvd