SpikeInterface/spikeinterface #840: saving `DeepInterpolatedRecording`

khl02007, 2022-07-22:
@alejoe91 I'm unable to save a deepinterpolated recording with multiple jobs. getting this error:
`tensorflow/compiler/mlir/tools/kernel_gen/tf_gpu_runtime_wrappers.cc:30] 'cuModuleUnload(it.second)' failed with 'CUDA_ERROR_NOT_INITIALIZED'`. any clues?

[image]

khl02007, 2022-07-26:
Here is what I'm using in case it helps:

```
ubuntu 20.04
nvidia driver 510
cuda=11.7
cudatoolkit=11.2 
cudnn=8.1.0
tensorflow=2.9
```

can save with `n_jobs=1` but when using `n_jobs` greater than 1, it just hangs

alejoe91, 2022-07-26:
Thanks for the additional info @khl02007 

Just to be sure, if you use `use_gpu=False` at instantiation, then the parallel processing runs fine?

khl02007, 2022-07-27:
Hmm, actually even when `use_gpu=False` the parallel saving hangs - so maybe this is a parallelization issue rather than a GPU issue

khl02007, 2022-07-27:
Parallel saving of other types of recordings (e.g. binary recording) works fine

khl02007, 2022-07-28:
@alejoe91 another Q: for the parallel jobs with specified chunk duration, does it mean that each chunk is treated separately? that may be problematic for the edges where the chunk doesn't contain complete information to infer. in other words there should be overlaps between adjacent chunks and this should be `n_frames_before`  and `after` frames long.

alejoe91, 2022-08-02:
@khl02007 can you try to add the argument: `mp_context="spawn"` to the `save()` call? 

The extra frames are taken care of [here](https://github.com/SpikeInterface/spikeinterface/blob/678a31eca474027e7d8497e4f2bc7c8746bb2ee1/spikeinterface/preprocessing/deepinterpolation/deepinterpolation.py#L82-L100), since the "chunk" doesn't know it's a chunk!

alejoe91, 2022-08-02:
BTW, that's on linux or Mac?

khl02007, 2022-08-02:
> @khl02007 can you try to add the argument: `mp_context="spawn"` to the `save()` call?

Yes, that works! thanks 

> The extra frames are taken care of [here](https://github.com/SpikeInterface/spikeinterface/blob/678a31eca474027e7d8497e4f2bc7c8746bb2ee1/spikeinterface/preprocessing/deepinterpolation/deepinterpolation.py#L82-L100), since the "chunk" doesn't know it's a chunk!

oh good point! thanks for resolving this issue.

closed 2022-08-02 by khl02007