SpikeInterface/spikeinterface #841: No module named \'spikeinterface\'\n'

phanhuynh, 2022-07-25:
Hello team, 
I've just git pulled the latest version of spikeinterface and set the SPIKEINTERFACE_DEV_PATH.  
I'm getting the following error message:
```
SpikeSortingError: Spike sorting in docker failed with the following error:
b'Traceback (most recent call last):\n  File "/Users/huynh/spike_epsztein/proj_louis_2/in_container_sorter_script.py", line 3, in <module>\n    from spikeinterface import load_extractor\nModuleNotFoundError: No module named \'spikeinterface\'\n'
```
Thank you so much in advance!

vncntprvst, 2022-07-29:
Hi @phanhuynh 
Could you provide a bit more detail about the commands and parameters leading to this error ? 
Thanks

phanhuynh, 2022-07-30:
Hello Vincent,

Thank you for your response.  Unfortunately, I'll be out of my office until
mid August.

However, Alessio and I have been talking about this before so, hopefully,
he will be aware of the context.

Thank you!

phanhuynh, 2022-08-19:
Hi Alessio,

I hope you've been well.  I'm curious if you have any news on this?

Thank you so much!

vncntprvst, 2022-08-20:
Hi Phan,
I've been able to reproduce that bug, but I haven't finished fixing it yet. I'll get back to it this week. 
Vincent

phanhuynh, 2022-09-06:
Hello Vincent,

I hope you've been well.  I just wanted to check in on your progress.  It
would be great to be able to use Kilosort3.

Thank you so much!