SpikeInterface/spikeinterface #844: Request: Explicit installation instructions

TdSNielsen, 2022-07-26:
Dear developers,
(I'm not sure if this should be an issue or a discussion)

this may be because I'm new to Python (and a bit dull), but I have had a lot of frustrations over getting all the dependencies to work over the last couple of months... I have tried to reinstall a couple of times, and have tried different approaches to installing, but can't quite get everything to work (at the moment struggling with the sorters). I'm wondering if it could be because Python is very sensitive to which version you have of all the packages (I have been installing the latest version, which might be wrong), which compiler you're using (I'm using PyCharm, from examples it seems like you're using Jupyter Notebook) and to the operating system and hardware (I'm mainly using a Mac M1pro, but also a Windows desktop). 
I am also a little confused about whether it is supposed to be possible to call the sorters without installing them as separate packages (docker images), so far, I have not been able to call the sorting functions without installing a package for the sorter I'm calling and I have only been able to get tridesclous and klusta to work...

I was therefore wondering if you could publish a step-by-step guide (for dummies ;-)), detailing how to get a specific setup to work? - This could, e.g., be how to set up everything for your eLIFE paper from installing the Python and Anaconda environments, compiler, required packages, SpikeInterface itself, and sorters etc. (if necessary)...

It would also be helpful with a bit of explicit information/discussion of hardware limitations: My initial thought was to experiment with SpikeInterface on my Mac, then potentially transfer to the university's (Linux-based) server once I got a pipeline working, but this might not be possible if there are hardware limitations (e.g., CUDA would be a problem on both my mac and Linux)...

Sorry if it is just me that is being to thick-headed to understand this (but I do imagine that there are many beginners interested in using spike sorting tools). I do have programming experience and find the Python language fairly accessible, but package handling is rather frustrating.

Best,
Thomas

alejoe91, 2022-08-02:
@TdSNielsen thanks for your suggestion! I definitely agree that a step-by-step guide would be helpful! We are planning a release in September and we'll do our best to accompany it with an enhancement of the docs, including a setup guide "for dummies" ;)

samuelgarcia, 2022-08-22:
Hi,
we already have this for user with few knowledge on python ecosystem.
https://github.com/SpikeInterface/spikeinterface/tree/master/installation_tips
The "quick installation" generally works for students in my lab that never install python on there machine.

alejoe91, 2022-10-13:
@TdSNielsen we're closing this now, but we'd be happy to discuss more about how to improve documentation on our side!

closed 2022-10-13 by alejoe91