SpikeInterface/spikeinterface #850: memory usage problems in latest dev version using NWB

lfrank, 2022-07-27:
We're experiencing much slower recording creation in the latest version and much increased memory usage.

The raw data are in NWB, and extracting ~90 minutes from a single tetrode ends up requiring > 80GB of RAM, even though the total memory was specified as 1G.
It also seems to take much long than it did a month or two ago as far as we can tell. 

Any thoughts would be appreciated.

lfrank, 2022-07-29:
@bendichter @alejoe91  Is this something either of you would have time to look at? I tried looking through the recording.save function and it wasn't obvious to me where one might, for example, delete old chunks to save RAM and/or make things more efficient.

I should add that we're using only one job here because in the past it didn't seem to help to increase the number of jobs.

alejoe91, 2022-08-01:
Hi @lfrank 

Could you share one of the datasets?

lfrank, 2022-08-01:
> This Message Is From an External Sender
> This message came from outside your organization.

@alejoe Thanks for checking in on this. We just discovered that it has to do with the h5py version: 2.10 is fast; 3.7 is slow and uses much more memory. I’ll pass this on to the pynwb people.

alejoe91, 2022-08-01:
Closing then!

closed 2022-08-01 by alejoe91

bendichter, 2022-08-02:
@lfrank Interesting! We'll have to look into that

lfrank, 2022-08-02:
As a follow up, we tracked this down to the version of h5py: 2.10 works well, 3.5, 3.6, and 3.7 all use 5-6x RAM and take 2-3x as long to write the recording.save.

For the moment our plan was to restrict our environment to version 2.10, but obviously that’s not a good long term solution.

ZjQcmQRYFpfptBannerStart

ZjQcmQRYFpfptBannerEnd