SpikeInterface/spikeinterface #851: bug using mountainsort4 in new version

lfrank, 2022-07-27:
Running the latest dev version and trying to use the latest MountainSort4 (1.0.2) from pypi gives an error:

```
Mountainsort4 use the OLD spikeextractors mapped with NewToOldRecording
Traceback (most recent call last):
  File "/Users/loren/opt/anaconda3/envs/spyglass/lib/python3.8/site-packages/spikeinterface/sorters/basesorter.py", line 225, in run_from_folder
    SorterClass._run_from_folder(output_folder, sorter_params, verbose)
  File "/Users/loren/opt/anaconda3/envs/spyglass/lib/python3.8/site-packages/spikeinterface/sorters/mountainsort4/mountainsort4.py", line 123, in _run_from_folder
    if parse(mountainsort4.__version__) >= parse("1.0.2"):
AttributeError: module 'mountainsort4' has no attribute '__version__'
```
@magland One easy solution would of course be to include __version__, but I don't know if that's how this should be solved.

magland, 2022-08-01:
Should be fixed in mountainsort4 1.0.3

alejoe91, 2022-08-02:
@lfrank working on my side!

closed 2022-08-02 by alejoe91