SpikeInterface/spikeinterface #858: `has_scaled_traces()` method renamed to `has_scaled`

h-mayorquin, 2022-08-01:
I don't know if it was intended but one the commits in the last two days (Today being 2022-01-01) changed the name of `has_scaled_traces` to `has_scaled` in the master branch and broke our stuff in neuroconv:

https://github.com/catalystneuro/neuroconv/pull/45

I think is this new re-organization around the concept of snippets. This is the new function which also seems repeated and with a new name:
https://github.com/SpikeInterface/spikeinterface/blob/678a31eca474027e7d8497e4f2bc7c8746bb2ee1/spikeinterface/core/baserecordingsnippets.py#L49-L64

Maybe I am getting something wrong.

alejoe91, 2022-08-01:
Hi @h-mayorquin 

I though I left the `has_scaled_traces()` for back compatibility, but I'll double check! In case it's not there we'll re-add it. 

The rationale is that since now the BaseSnippets and BaseRecording inherit from the same class, the semantic of `has_scaled_traces()` only applies to the Recording, while `has_scaled()` is more general. Let's discuss about it tomorrow

h-mayorquin, 2022-08-01:
Yes, I think you accidentally removed it from `BaseRecording`.

PR #859 (2022-08-02): Recover has scaled traces
This should close #858 

I also eliminated a duplication of `has_scales` one duplication in the same page in baserecordingsnipets and one duplication (as it is already inherited) on basesnipets.

I added the cleanup of segutils here as well
closed 2022-08-02 by alejoe91