SpikeInterface/spikeinterface #871: Error "bad record dict" occurred when running "ss.run_sorters"

betty-hbr, 2022-08-05:
Hi all, I’m running 0.90.1 SpikeInterface with python 3.8.13, dataset was recorded using tetrode probes. And I encountered this error.
    
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
Input In [57], in <cell line: 3>()
      1 # run spike sorting on entire recording
      2 working_folder = '/Users/betty_hbr/Dropbox/1_Project3/Workspace/results_MS4'
----> 3 sorting_MS = ss.run_sorters('mountainsort4', recording_rb, working_folder=working_folder)
      4 print('Found', len(sorting_MS.get_unit_ids()), 'units')

File ~/Applications/anaconda3/envs/si090/lib/python3.8/site-packages/spikeinterface/sorters/launcher.py:142, in run_sorters(sorter_list, recording_dict_or_list, working_folder, sorter_params, mode_if_folder_exists, engine, engine_kwargs, verbose, with_output)
    140     recording_dict = recording_dict_or_list
    141 else:
--> 142     raise ValueError('bad recording dict')
    144 need_dump = engine != 'loop'
    145 task_args_list = []

ValueError: bad recording dict

alejoe91, 2022-08-05:
Hi!

How did you get `recording_rb`? Can you paste the output of: `recording_rb.to_dict()`?

betty-hbr, 2022-08-05:
Hi 
I get `recording_rb` as a result of `recording_rb = st.remove_bad_channels(recording_cmr, bad_threshold=5)`. 
Here is the output of `recording_rb.to_dict()`:

{'class': 'spikeinterface.toolkit.preprocessing.remove_bad_channels.RemoveBadChannelsRecording',
 'module': 'spikeinterface',
 'kwargs': {'recording': {'class': 'spikeinterface.toolkit.preprocessing.common_reference.CommonReferenceRecording',
   'module': 'spikeinterface',
   'kwargs': {'recording': {'class': 'spikeinterface.toolkit.preprocessing.filter.BandpassFilterRecording',
     'module': 'spikeinterface',
     'kwargs': {'recording': {'class': 'spikeinterface.core.channelslicerecording.ChannelSliceRecording',
       'module': 'spikeinterface',
       'kwargs': {'parent_recording': {'class': 'spikeinterface.core.numpyextractors.NumpyRecording',
         'module': 'spikeinterface',
         'kwargs': {'traces_list': [array([[  0.        ,   0.        ,   0.        , ...,   0.        ,
                     0.        ,   0.        ],
                  [  3.136     ,   5.7599998 , -16.447998  , ..., -16.255999  ,
                   -26.687998  ,  -8.447999  ],
                  [  6.6559997 ,   0.12799999,  -1.0239999 , ...,  -9.087999  ,
                   -19.583998  , -10.176     ],
                  ...,
                  [ -4.6079993 ,  -3.3279998 ,   3.7759998 , ...,  -2.2399998 ,
                    -6.9119997 ,  -3.008     ],
                  [  9.215999  ,   5.1199994 ,   4.2239995 , ...,  -3.968     ,
                    -3.3279998 ,   2.1119998 ],
                  [  6.272     ,  10.687999  ,  10.1119995 , ...,   2.1119998 ,
                     6.6559997 ,   4.9919996 ]], dtype=float32)],
          'sampling_frequency': 24414},
         'dumpable': False,
         'version': '0.90.1',
         'relative_paths': False,
         'annotations': {'is_filtered': False},
         'properties': {'group': None,
          'location': None,
          'gain_to_uV': None,
          'offset_to_uV': None}},
        'channel_ids': range(0, 16),
        'renamed_channel_ids': range(0, 16)},
       'dumpable': True,
       'version': '0.90.1',
       'relative_paths': False,
       'annotations': {'is_filtered': False},
       'properties': {'group': array([0, 0, 0, 0, 1, 1, 1, 1, 2, 2, 2, 2, 3, 3, 3, 3]),
        'location': array([[ 1.0000000e+01,  0.0000000e+00],
               [ 6.1232340e-16,  1.0000000e+01],
               [-1.0000000e+01,  1.2246468e-15],
               [-1.8369702e-15, -1.0000000e+01],
               [ 1.0000000e+01,  3.6000000e+02],
               [ 6.1232340e-16,  3.7000000e+02],
               [-1.0000000e+01,  3.6000000e+02],
               [-1.8369702e-15,  3.5000000e+02],
               [ 3.7000000e+02,  0.0000000e+00],
               [ 3.6000000e+02,  1.0000000e+01],
               [ 3.5000000e+02,  1.2246468e-15],
               [ 3.6000000e+02, -1.0000000e+01],
               [ 3.7000000e+02,  3.6000000e+02],
               [ 3.6000000e+02,  3.7000000e+02],
               [ 3.5000000e+02,  3.6000000e+02],
               [ 3.6000000e+02,  3.5000000e+02]]),
        'gain_to_uV': None,
        'offset_to_uV': None}},
      'freq_min': 300,
      'freq_max': 6000,
      'margin_ms': 5.0},
     'dumpable': True,
     'version': '0.90.1',
     'relative_paths': False,
     'annotations': {'is_filtered': True},
     'properties': {'group': array([0, 0, 0, 0, 1, 1, 1, 1, 2, 2, 2, 2, 3, 3, 3, 3]),
      'location': array([[ 1.0000000e+01,  0.0000000e+00],
[... 43 lines cut ...]
    'offset_to_uV': None}},
  'bad_threshold': 5},
 'dumpable': True,
 'version': '0.90.1',
 'relative_paths': False,
 'annotations': {'is_filtered': True},
 'properties': {'group': array([0, 0, 0, 0, 1, 1, 1, 1, 2, 2, 2, 2, 3, 3, 3, 3]),
  'location': array([[ 1.0000000e+01,  0.0000000e+00],
         [ 6.1232340e-16,  1.0000000e+01],
         [-1.0000000e+01,  1.2246468e-15],
         [-1.8369702e-15, -1.0000000e+01],
         [ 1.0000000e+01,  3.6000000e+02],
         [ 6.1232340e-16,  3.7000000e+02],
         [-1.0000000e+01,  3.6000000e+02],
         [-1.8369702e-15,  3.5000000e+02],
         [ 3.7000000e+02,  0.0000000e+00],
         [ 3.6000000e+02,  1.0000000e+01],
         [ 3.5000000e+02,  1.2246468e-15],
         [ 3.6000000e+02, -1.0000000e+01],
         [ 3.7000000e+02,  3.6000000e+02],
         [ 3.6000000e+02,  3.7000000e+02],
         [ 3.5000000e+02,  3.6000000e+02],
         [ 3.6000000e+02,  3.5000000e+02]]),
  'gain_to_uV': None,
  'offset_to_uV': None}}

alejoe91, 2022-08-05:
Ok, the problem is trhe following: you are using a `NumpyRecording` which is an in-memory recording that cannot be serialized.

In order to run the `run_sorters()` function, recordings need to be "saved" to a json file, but in-memory object son't support this. 

So, you have 2 alternatives:
1. Save your recording to a binary object: `recording_rb = recording_rb.save(folder="my-folder", chunk_size=30000, progress_bar=True)`. This saves the in-memory object to a binary file that can be serialized.
2. Instead of using the `run_sorters()` function, use the `run_sorter()` to run a single sorter.

I recommend option 1, since it's nice to have a recording bound to a file! Let me know if it works!

Alessio

betty-hbr, 2022-08-08:
Yes, it perfectly solved the problem !!!

Many thanks!!!

closed 2022-08-08 by alejoe91