SpikeInterface/spikeinterface #874: Export quality metrics from spikeinterface to phy

tinaxia2016, 2022-08-05:
Hi there, I'm a new user of the spikeinterface. I'm wondering if there is any example code for exporting the quality metrics from spikeinterface to phy. Being able to look at the quality metrics in the cluster view of phy will help the manual sorting step a lot. 

Thanks in advance!

colehurwitz, 2022-08-10:
@alejoe91 Is there a way to do this? I couldn't figure it out.

alejoe91, 2022-08-11:
Hi guys,

If you have computed quality metrics (so `quality_metrics` is available as an extension), then they should be epxorted automatically as .tsv files. See: https://github.com/SpikeInterface/spikeinterface/blob/master/spikeinterface/exporters/to_phy.py#L228-L237

Is it not working?

colehurwitz, 2022-08-11:
@tinaxia2016 I think this is how it works!

colehurwitz, 2022-08-11:
Thanks @alejoe91 👍

alejoe91, 2022-08-12:
@tinaxia2016 

To clarify: quality metrics are automatically exported to phy if you do the following before exporting.
```
qm = si.compute_quality_metrics(we)
```
where `we` is your `WaveformExtractor`. Internally, this makes a waveform extension called `quality_metrics`, meaning that you have a "quality_metrics" folder in your waveform folder that can be used by other functions (and reloaded when loading from folder)

closed 2022-08-12 by alejoe91

alejoe91, 2022-08-12:
1. The AP stream from spikeGLX is already high pass filtered, but sometimes channels have an offset so imo it's better to refilter 

2. I think that the mismatch is due to a different sparsity (the channels associated to each unit). In the ks sparsity is using the geometry, while by default SI uses the channels with the largest amplitude. If you have some noisy channels (in your case the bottom ones seem noisy), they kind of mess things up. You can choose another sparsity option using the radius by adding the argumend: `sparsity_dict=dict(method='radius', radius_um=100)`. I think we should change the default sparsity to this! You can also try to increase `max_spikes_per_unit` to 1000 or 2000 to get a better estimate.

(BTW that unit is clearly noise given the very high ISI violation ratio!)

tinaxia2016, 2022-08-12:
Thanks @alejoe91! I'm not sure whether this unit is clearly noise. I thought it was a good unit judging by the dip in the center of ACG. What's the acceptable range for ISI violation ratio?

alejoe91, 2022-08-12:
We usually remove everything above 0.5, but I agree the ACG looks okayish! Maybe you can try to clean it up a bit in phy ;)