SpikeInterface/spikeinterface #880: Subsampled waveform looks weird in Phy

tinaxia2016, 2022-08-12:
Ideally, I would like to calculate WaveformExtractor "we" without setting a limit to "max_spikes_per_unit". But it requires the allocation of a large memory which exceeds my RAM. So I didn't change the default setting (max_spike_per_unit=500). Then I found that the waveform looks weird in Phy. Is this because of the subsampling on waveforms?  

The code I use to calculate WaveformExtractor:
```
sorting = se.read_phy('the folder path with Kilosort output to Phy', exclude_cluster_groups='noise')

recording = se.SpikeGLXRecordingExtractor('the folder path with neuropixel recording', stream_id='imec0.ap')
recording_f = st.bandpass_filter(recording, freq_min=300, freq_max=6000)

folder = './Official_Tutorial_SI_0.94_May2022/waveform'
we = si.extract_waveforms(recording_f, sorting, folder,
                          progress_bar=True,
                          n_jobs=4, total_memory="500M", overwrite=True, load_if_exists=False)
```

One example unit in Phy: 
[image]

The same unit in the original Phy output:
[image]

alejoe91, 2022-09-09:
Hi @tinaxia2016 

Sorry for our late reply. This is probably an issue with the sparsity assigned to each unit, which is the subset of channels on which the unit is visualized. You see that the channels do not correspond at all. 
The default way that the sparsity is computed, is by taking the N channels with the largest amplitude. If you have very noisy channels, this can lead to weird results.

How does the template look like when you plot it in SpikeInterface?

`sw.plot_unit_templates(we, unit_ids=[your-unit-id])`

tinaxia2016, 2022-10-17:
Sorry for my late reply. Here is what I got with plot_unit_templates: 
[image]

alejoe91, 2022-12-28:
@tinaxia2016 can you try to zoom in and check the template?

Also sorry for my late reply!

alejoe91, 2023-03-27:
Closing for inactivity. Feel free to reopen if the errir still persists