SpikeInterface/spikeinterface #885: Container using system packages

JuliaSprenger, 2022-08-17:
When running a sorter using `run_sorter_container` some system volumens are mounted. On ubuntu and using singularity this can lead to the system installation of spikeinterface (e.g. in `~.local/pythonX.X/site-packages`) to shadow the spikeinterface version installed in the container.
This can lead to confusing version conflicts, but can also be used to provide the container with a specific spikeinterface version. Is this an intended feature or a bug? In the first case it might be good to add a warning if the container internal spikeinterface version is shadowed by the system version.

samuelgarcia, 2022-08-22:
Hi Julia,
when spikeinterface is already in the container, we should not have new installtion on the fly.
https://github.com/SpikeInterface/spikeinterface/blob/master/spikeinterface/sorters/runsorter.py#L468

regarding the logic, maybe if you have an old spikeinterface version, then a new one is installed.

chyumin, 2022-09-12:
@JuliaSprenger Would you mind to test changes in #938 ?

samuelgarcia, 2022-09-13:
Hi Yu Min, 
I am OK with the patch you did.
What I do not understand in that case of Julia is why we try to install spikeinterface package in the container when it is already installed. This should not happen.
Am I missing something ?

chyumin, 2022-09-13:
Hi @samuelgarcia 
I don't think we try to install

JuliaSprenger, 2022-09-13:
In general we install spikeinterface in the `run_sorter_in_container` function if it's not present in the container or there's an issue with the version installed in the container. In the context of this issue I tried to install a specific version as sometimes the sorters are not running with different spikeinterface versions outside and inside the container.

chyumin, 2022-09-13:
I believe we don't have spikeinterface installed in any docker image, so it should always try to install, no?

What happens is that one of the system volumes that is binded in singularity mode is the host's `$HOME` folder, so one of the paths that Python searches for (and installs) packages is `/home/<user>/.local/pythonX.X/site-packages`, so when running in singularity mode it turns out that Python inside container also shares this folder with the host. Which I believe is not desired and also doesn't happen in docker mode

PR #938 (2022-10-05): Python user base folder
`PYTHONUSERBASE` ([reference](https://docs.python.org/2/using/cmdline.html#envvar-PYTHONUSERBASE)) prevents Singularity from using, creating or overwriting packages on host's `~/.local/lib/python3.X/site-packages` folder.

fixes #885
closed 2022-10-05 by alejoe91