SpikeInterface/spikeinterface #887: Can't dump NumpyRecording extractor

TomBugnon, 2022-08-18:
Hi,

When running `run_sorter` with a recording issued from a `NumpyRecording` object I get the following Error

```ss.run_sorter(sorter, si_recording, output_folder=kilosort_output_dir, verbose=True, **params)```

> File ~/miniconda3/envs/ecephys_tdt_dev/lib/python3.10/site-packages/spikeinterface/sorters/runsorter.py:67, in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, singularity_image, with_output, **sorter_params)
>      61 if singularity_image is not None:
>      62     return run_sorter_container(sorter_name, recording, 'singularity', singularity_image,
>      63                                 output_folder=output_folder,
>      64                                 remove_existing_folder=remove_existing_folder,
>      65                                 delete_output_folder=delete_output_folder, verbose=verbose,
>      66                                 raise_error=raise_error, with_output=with_output, **sorter_params)
> ---> 67 return run_sorter_local(sorter_name, recording, output_folder=output_folder,
>      68                         remove_existing_folder=remove_existing_folder,
>      69                         delete_output_folder=delete_output_folder,
>      70                         verbose=verbose, raise_error=raise_error, with_output=with_output,
>      71                         **sorter_params)
> 
> File ~/miniconda3/envs/ecephys_tdt_dev/lib/python3.10/site-packages/spikeinterface/sorters/runsorter.py:84, in run_sorter_local(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, **sorter_params)
>      81 SorterClass = sorter_dict[sorter_name]
>      83 # only classmethod call not instance (stateless at instance level but state is in folder)
> ---> 84 output_folder = SorterClass.initialize_folder(
>      85     recording, output_folder, verbose, remove_existing_folder)
>      86 SorterClass.set_params_to_folder(
>      87     recording, output_folder, sorter_params, verbose)
>      88 SorterClass.setup_recording(recording, output_folder, verbose=verbose)
> 
> File ~/miniconda3/envs/ecephys_tdt_dev/lib/python3.10/site-packages/spikeinterface/sorters/basesorter.py:117, in BaseSorter.initialize_folder(cls, recording, output_folder, verbose, remove_existing_folder)
>     115 rec_file = output_folder / 'spikeinterface_recording.json'
>     116 if recording.is_dumpable:
> --> 117     recording.dump_to_json(rec_file)
>     118 else:
>     119     d = {'warning': 'The recording is not dumpable'}
> 
> File ~/miniconda3/envs/ecephys_tdt_dev/lib/python3.10/site-packages/spikeinterface/core/base.py:457, in BaseExtractor.dump_to_json(self, file_path, relative_to, folder_metadata)
>     450 dump_dict = self.to_dict(include_annotations=True,
>     451                          include_properties=False,
>     452                          include_features=False,
>     453                          relative_to=relative_to,
>     454                          folder_metadata=folder_metadata)
>     455 file_path = self._get_file_path(file_path, ['.json'])
>     456 file_path.write_text(
> --> 457     json.dumps(check_json(dump_dict), indent=4),
>     458     encoding='utf8'
>     459 )
> 
> File ~/miniconda3/envs/ecephys_tdt_dev/lib/python3.10/json/__init__.py:238, in dumps(obj, skipkeys, ensure_ascii, check_circular, allow_nan, cls, indent, separators, default, sort_keys, **kw)
>     232 if cls is None:
>     233     cls = JSONEncoder
>     234 return cls(
>     235     skipkeys=skipkeys, ensure_ascii=ensure_ascii,
>     236     check_circular=check_circular, allow_nan=allow_nan, indent=indent,
>     237     separators=separators, default=default, sort_keys=sort_keys,
> --> 238     **kw).encode(obj)
> 
> File ~/miniconda3/envs/ecephys_tdt_dev/lib/python3.10/json/encoder.py:201, in JSONEncoder.encode(self, o)
>     199 chunks = self.iterencode(o, _one_shot=True)
>     200 if not isinstance(chunks, (list, tuple)):
> --> 201     chunks = list(chunks)
>     202 return ''.join(chunks)
> 
> File ~/miniconda3/envs/ecephys_tdt_dev/lib/python3.10/json/encoder.py:431, in _make_iterencode.<locals>._iterencode(o, _current_indent_level)
>     429     yield from _iterencode_list(o, _current_indent_level)
>     430 elif isinstance(o, dict):
> --> 431     yield from _iterencode_dict(o, _current_indent_level)
>     432 else:
>     433     if markers is not None:
> 
> File ~/miniconda3/envs/ecephys_tdt_dev/lib/python3.10/json/encoder.py:405, in _make_iterencode.<locals>._iterencode_dict(dct, _current_indent_level)
[... 63 lines cut ...]
    455         file_path = self._get_file_path(file_path, ['.json'])
    456         file_path.write_text(
--> 457             json.dumps(check_json(dump_dict), indent=4),
    458             encoding='utf8'
    459         )

And here is the content of the `dump_dict`
> {'class': 'spikeinterface.core.segmentutils.ConcatenateSegmentRecording', 'module': 'spikeinterface', 'kwargs': {'recording_list': [{'class': 'spikeinterface.core.numpyextractors.NumpyRecording', 'module': 'spikeinterface', 'kwargs': {'traces_list': [array([[-2.7660799e-04,  2.6111999e-05, -3.2415998e-04, ...,
>         -1.1244800e-04, -1.3695999e-04, -1.0604799e-04],
>        [-2.6495999e-04,  3.3663997e-05, -3.2825596e-04, ...,
>         -8.4159990e-05, -1.0393599e-04, -1.0681599e-04],
>        [-2.7039996e-04,  2.2143999e-05, -3.3766398e-04, ...,
>         -9.5039992e-05, -1.3951999e-04, -9.4783994e-05],
>        ...,
>        [-2.5919999e-05,  2.4959998e-05, -3.7887996e-05, ...,
>          4.9471997e-05,  1.5167999e-05, -8.8959996e-06],
>        [-1.5295998e-05,  2.1823998e-05, -2.2399998e-05, ...,
>          4.2239997e-05, -5.0559997e-06,  1.4271999e-05],
>        [-2.4127998e-05,  5.1839997e-06, -4.6015997e-05, ...,
>          5.4719996e-05, -5.3759995e-06,  1.9455998e-05]], dtype=float32)], 't_starts': None, 'sampling_frequency': 24414.0625}, 'dumpable': False, 'version': '0.94.0', 'relative_paths': False, 'annotations': {'is_filtered': False}, 'properties': {'group': [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], 'location': [[0.0, 800.0], [0.0, 750.0], [0.0, 700.0], [0.0, 650.0], [0.0, 600.0], [0.0, 550.0], [0.0, 500.0], [0.0, 450.0], [0.0, 400.0], [0.0, 350.0], [0.0, 300.0], [0.0, 250.0], [0.0, 200.0], [0.0, 150.0], [0.0, 100.0], [0.0, 50.0]], 'gain_to_uV': None, 'offset_to_uV': None}}], 'ignore_times': True}, 'dumpable': True, 'version': '0.94.0', 'relative_paths': False, 'annotations': {'is_filtered': False, 'probe_0_planar_contour': [[-10, 50], [0, 0], [10, 50], [40, 850], [-40, 850]]}, 'properties': {'group': [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], 'location': [[0.0, 800.0], [0.0, 750.0], [0.0, 700.0], [0.0, 650.0], [0.0, 600.0], [0.0, 550.0], [0.0, 500.0], [0.0, 450.0], [0.0, 400.0], [0.0, 350.0], [0.0, 300.0], [0.0, 250.0], [0.0, 200.0], [0.0, 150.0], [0.0, 100.0], [0.0, 50.0]], 'gain_to_uV': None, 'offset_to_uV': None}}

Setting `recording.is_dumpable = False` fixes it
For the record I'm using a NumpyRecording because the TDT extractor is broken.

Best, Tom

TomBugnon, 2022-08-18:
Since we're at it, are NumpyRecordings supposed to be sortable? 

I get this error from Kilsort2_5 when intending to sort one

> --------------------------------------------------------------------------------Error using gpuArray/subsref
> 
> Subscript indices must either be real positive integers or logicals.Error using gpuArray/subsref
> 
> Subscript indices must either be real positive integers or logicals.

Which is apparently indicative of kilosort not finding any spikes, either because the channel locations or the binary file are messed up.
I'm wondering if `write_recording_as_binary` works properly for NumpyRecording because the shape of the traces is (nsamples x nchans) while I gather Kilosort would expect a (nchans x nsamples) binary file?

samuelgarcia, 2022-08-22:
Yes NumpyRecording can be sorted be very often sorter need a binary file as input so very often an intermediate file is written to the disk. This is the case for kilosort.
kilosort need a (nsample, nchans) shape, so the write_recording_as_binary should be OK.

The best way to avoid problems when you have a numpy recording is to "save" it so internally it will be binary file.

`rec2 = rec.save(folder="/path/to/somewher", n_jobs=1, chunk_size=10000, progress_bar=True)`

then rec2 will be dumpable and kilosort will not try to convert it again to a file.