SpikeInterface/spikeinterface #888: Spikeinterface-gui not working and not imported

PaolaGith, 2022-08-19:
Hello,
I have issues with spikeinterface-gui starting from the import.

As described in the tutorial, I import 
import spikeinterface_gui as sigui 
in my jupyter notebook script.
But I get this error 

ModuleNotFoundError                       Traceback (most recent call last)
Input In [3], in <cell line: 6>()
      3 from pathlib import Path
      5 import ephyviewer
**----> 6 import spikeinterface_gui as sigui**
      8 # %matplotlib widget
      9 get_ipython().run_line_magic('matplotlib', 'inline')

File ~/anaconda3/envs/si_env/lib/python3.9/site-packages/spikeinterface_gui/__init__.py:8, in <module>
      5 from .version import version as __version__
      7 from .myqt import QT, mkQApp
----> 8 from .mainwindow import MainWindow
      9 from .controller import SpikeinterfaceController
     11 # views

File ~/anaconda3/envs/si_env/lib/python3.9/site-packages/spikeinterface_gui/mainwindow.py:6, in <module>
      1 import time
      3 from .myqt import QT, QT_MODE
----> 6 from .controller import SpikeinterfaceController
      8 from .viewlist import possible_class_views
     11 class MainWindow(QT.QMainWindow):

File ~/anaconda3/envs/si_env/lib/python3.9/site-packages/spikeinterface_gui/controller.py:7, in <module>
      4 from .myqt import QT
      6 from spikeinterface.widgets.utils import get_unit_colors
----> 7 from spikeinterface.postprocessing import (WaveformPrincipalComponent, get_template_extremum_channel, 
      8                                            get_template_channel_sparsity, compute_correlograms, compute_unit_locations,
      9                                            compute_template_similarity)
     10 from spikeinterface.qualitymetrics import compute_num_spikes
     12 import numpy as np

ModuleNotFoundError: No module named 'spikeinterface.postprocessing'

Although I installed all the packages, in my conda environment I do not have spikeinterface_gui installed, but spikeinterface-gui.
Can the different spelling when importing be the problem?
Nevertheless, if I try to import spikeinterface-gui, it gives a syntax error. So the only way for me to move on with the sorting, is to comment this line.

I started to get this problem recently. Before I was still not able to use the gui but at least I did not get any error during the import.

Can you please help?

Thank you

samuelgarcia, 2022-08-22:
Hi,
you should install sigui from source.
I will try to make a release soon.

samuelgarcia, 2022-10-13:
No it is released.
Upgrade bith package an dit should work.