SpikeInterface/spikeinterface #892: Why does Combinato give me different results in SpikeInterface compared to from the terminal?

dustinf1989, 2022-08-19:
I recently found out about SpikeInterface and think it is a great idea to compare Spike Sorting algorithms. I was using Combinato from the terminal (how it is described on its own GitHub) going through the [tutorial III ](https://github.com/jniediek/combinato/wiki/Tutorial-Real-Data) with the supplied file, CSC29.ncs. I then completed the same pipeline in SpikeInterface, but I get slightly different results and I'm not sure why.  I have 2 extra clusters when using SpikeInterface and the waveform amplitude is much larger. It seems some kind of scaling factor is missing in SpikeInterface. Additionally, I was wondering how and if SpikeInterface recognized Combinato's "options" file? Combinato's options file has more keys than the default options in SpikeInterface, so I was wondering if this could be leading to the discrepency.

The images on the left are from running Combinato in SpikeInterface and scaled to 200uV while those on the right are from running the Combinato tutorial and scaled to 130uV.
[image]

ferchaure, 2022-08-19:
Hi dustin

> it seems some kind of scaling factor is missing in SpikeInterface

You are right, I will find a proper way to fix it and tag this issue. For the time being you can add this code to your pipeline:

```
from spikeinterface.toolkit.preprocessing import  ScaleRecording
fixed_recording = ScaleRecording(recording)
```
And then run combinato using `fixed_recording `. The difference in results could be related to the scales, some steps in combinato are scale dependent. 

>  I was wondering how and if SpikeInterface recognized Combinato's "options" file?

Spikeinterface creates a local_options.py file with the parameters inside the folder, setting all  the available parameters. You can find [here ](https://github.com/SpikeInterface/spikeinterface/blob/master/spikeinterface/sorters/combinato/combinato.py#L43-L84) the parameters available for combinato .

Cheers,
Fernando

alejoe91, 2022-08-23:
Merged now! https://github.com/SpikeInterface/spikeinterface/pull/894

dustinf1989, 2022-08-24:
Hi Fernando,

Thank you very much for the quick answer, and thanks for merging with the related issue. I can use `ScaleRecording` but how can I know what amount Combinato natively scales the data with? 

I looked at the parameters for Combinato in SpikeInterface before and there are much fewer than the `options.py` file that comes with Combinato. Here are all of the parameters:

```
options = {
    # How many clusters can be selected at one temperature?
    'MaxClustersPerTemp': 5,  # default 5
    # How many spikes does a cluster need to be selected?
    'MinSpikesPerClusterMultiSelect': 15,  # default 15

    # How many clustering recursions should be run?
    'RecursiveDepth': 1,  # default is 1 (do not recurse)

    # Iteratively recluster big clusters?
    'ReclusterClusters': True,
    # How many spikes does a cluster need to be re-clustered?
    'MinInputSizeRecluster': 2000,  # default 2000

    # How close do spikes have to be in the first template matching step?
    'FirstMatchFactor': .75,  # default .75

    # In the second template matching step?
    'SecondMatchFactor': 3,   # default 3

    # At what cluster distance does grouping stop?
    'MaxDistMatchGrouping': 1.8,   # default 1.8

    # Folder names the GUI should include
    'folder_patterns': ('CSC*', 'test', 'L??',
                        'R??', 'L???', 'R???', 'L????', 'R????',
                        'simulation_*','u*','F600*'),

    # Use this option to create raster plots in css-gui
    # This requires 'raster_options', see Documentation
    'RunGuiWithRaster': False,

    # Use this option to make css-gui read time stamps from
    # extraction thresholds instead of spike times (for display only)
    'GuiUseThresholdTimeAxis': False,
    'MinInputSize': 15,
    'TempStep': 0.01,
    'MarkArtifactClasses': True,
    'FractionOfBiggestCluster': .05,
    'nFeatures': 10,
    'Wavelet': 'haar',

    'ClusterPath': CLUS_BINARY,

    'ShowSPCOutput': False,
    'RecheckArtifacts': True,  # check after total match
    'plot': True,
    'plotTemps': True,
    'figsize': (2, 2),
    'dpi': 100,  # so figure is 200x200 pixels
    'ylim': (-200, 200),
    'linewidth': .4,
    'tempfigsize': (5, 5),
    'ArtifactOverview': False,
    'feature_factor': 3,
    'overwrite': True,
    'ExcludeVariableClustersMatch': True,
    'FirstMatchMaxDist': 4,   # default 4
    'SecondMatchMaxDist': 20,
    'OverwriteGroups': True,
    'smallmarker': 3,
    'bigmarker': 3,
    'max_amplitude': MAX_AMPLITUDE,
    'min_amplitude': MIN_AMPLITUDE,
    'compute_isi_upto_ms': 100,
    'isi_n_bins': 100,
    'isi_too_short_ms': 3,  # used to for isi plot, in single/multi unit crit
    'density_hist_bins': np.linspace(-MIN_AMPLITUDE,
                                     MAX_AMPLITUDE,
                                     MAX_AMPLITUDE),
    'cmap': cm.hot,
    'overview_ax_ylim': (-150, 150),
    'Debug': False,
    'histcolor': 'b',
    'histtype': 'stepfilled',
    'guistyle': 'oxygen',
    'UseCython': False
}

artifact_criteria = {
    'maxima': 5,
    'maxima_1_2_ratio': 2,
    'max_min_ratio': 1.5,
    'sem': 4,
    'ptp': 1
}
```

So what I'm wondering is, which values are used for the parameters in this file that aren't adjustable by SpikeInterface? Does SpikeInterface recognize the `options.py` file in the Combinato folder?

ferchaure, 2022-08-24:
> how can I know what amount Combinato natively scales the data with?

Combiato shouldn't scaled the data again, but it requires inputs as uV. Check the gains of your recording :

`rec_gains=recording1.get_channel_gains()
`

If the aren't None, add them as the gain argument in ScaleRecording(recording, gain=rec_gains)

> Does SpikeInterface recognize the options.py file in the Combinato folder

Yes, if the option isn't set on SpikeInterface, then the global options.py will be used. If you fine some option that could be relevant for reproducibility let me know and we can add it to the sorter parameters .

alejoe91, 2022-10-13:
Hi guys, can we close this?