SpikeInterface/spikeinterface #900: ModuleNotFoundError: No module named 'spikeinterface.postprocessing'

colehurwitz, 2022-08-22:
Hi all! I ran into an issue when installing spikeinterface. When I do a fresh install of spikeinterface:

`pip install spikeinterface`

I get an error when I do:

```python
from spikeinterface.extractors import NumpySorting
from spikeinterface.postprocessing import compute_correlograms
from spikeinterface.comparison import compare_two_sorters
from spikeinterface.widgets import plot_agreement_matrix
```
output:
`ModuleNotFoundError: No module named 'spikeinterface.postprocessing'`

This error goes away when I install spikeinterface locally (using `pip install -e .`). 

Is this the intended behaviour or a bug?

colehurwitz, 2022-08-22:
Maybe this is related to: #888 ?

alejoe91, 2022-08-23:
@colehurwitz the flattened modules `spikeinterface.postprocessing/preprocessing/qualitymetrics` are not released yet, so if you `pip install` you would still need the `toolkit` submodule. We are releasing within a couple of weeks :)

alejoe91, 2022-09-02:
@colehurwitz  we just released NEO, ProbeInterface, and SpikeInterface. Should be fixed if you do:

```
pip install -U neo
pip install -U probeinterface
pip install -U spikeinterface
```

closed 2022-10-13 by alejoe91