SpikeInterface/spikeinterface #905: Recent trouble loading OpenEphys data

BenATemple, 2022-08-24:
Have recently had trouble loading OpenEphys data. Maybe has something to do with the new OpenEphys GUI?
Is anyone else experiencing the same?
Screenshot attached
[image]
:

samuelgarcia, 2022-08-24:
Hi,
thanks for reporting this.
It is a bug in neo. The event (ttl) in your data is not managed correctly.
Could you open an issue in python-neo repo and also send us a small dataset that contain the same kind of data.

BenATemple, 2022-08-24:
Yes, absolutely. Sending you an email w/data.
Thanks for your help, let's get this fixed! I think this will be a problem for anyone with updated OpenEphys

alejoe91, 2022-08-24:
Can you send it to me too? :)

BenATemple, 2022-08-24:
Sure! What's your email?
Ben

BenATemple, 2022-08-24:
Yes! Just sent. Thanks a lot for your help guys

alejoe91, 2022-08-24:
@BenATemple do you have neo and spikeinterface installed from source? 

I can with no issue the the data you shared!

alejoe91, 2022-08-24:
In particular, I think it was fixed with this commit: https://github.com/NeuralEnsemble/python-neo/commit/51a167c3b9925c43cdb63a3e282a75a8d3270005

BenATemple, 2022-08-24:
ok awesome! let me check when I'm in lab -- maybe I need to update neo
I have the dev version of spike interface from last summer, maybe I need to
update that too?

BenATemple, 2022-08-24:
Hm, I uninstalled spike interface and reinstalled the most updated version. Still having trouble loading the same data that you guys were able to load ...
[image]
[image]

BenATemple, 2022-08-24:
Interesting ... if I just delete the "events" folder (which holds the TTL pulses), then I am able to load OK.
Good temporary workaround but still can't figure out what's going on on my end ...
Thanks a lot for your help!

alejoe91, 2022-08-25:
Try to `pip install --upgrade neo`

alejoe91, 2022-09-02:
@BenATemple we just released NEO, ProbeInterface, and SpikeInterface. Should be fixed if you do:

```
pip install -U neo
pip install -U probeinterface
pip install -U spikeinterface
```

BenATemple, 2022-09-28:
Forgot to confirm that this is fixed now -- thanks for your help!

closed 2022-09-28 by alejoe91