SpikeInterface/spikeinterface #906: How to load probe file / channel grouping for neuralynx data?

SusanL82, 2022-08-24:
Hello, 
I'm trying get spikeinterface working with neuralynx data and I'm running into some trouble when I try to get the tetrode grouping to work. I'm a little new to using python, so I apologize if this is a very obvious thing.

When I try to use load_probe_file, I get the error that "AttributeError: 'NeuralynxRecordingExtractor' object has no attribute 'load_probe_file'".
There's a similar error when I use get_sampling_frequency, so I guess that these options are just not possible for neuralynx data. So, how could I group my channels instead? If I just enter the contents of my .prb file, how can I add that to the recording? 

Thanks, 
Susan

os.environ['TEMPDIR'] = tempfile.gettempdir()
rec_folder = 'C:/Users/susan/Desktop/HD84/2019-12-19_16-09-01/'

########
recording = se.read_neuralynx(rec_folder)

chanIDs = recording.get_channel_ids()
num_chan = recording.get_num_channels()
num_segments = recording.get_num_segments()

print(f'Channel IDs: {chanIDs}')
print(f'sampling frequency: {fs}')
print(f'Nr of channels: {num_chan}')
print(f'Nr of segments: {num_segments}')

rec_prb = recording.load_probe_file('32TetPrb.prb')
Channel IDs: ['12' '29' '30' '31' '24' '25' '26' '27' '20' '21' '22' '13' '23' '16'
 '17' '18' '19' '4' '5' '6' '7' '0' '14' '1' '2' '3' '48' '49' '50' '51'
 '52' '53' '54' '15' '55' '32' '33' '34' '35' '36' '37' '38' '39' '40' '8'
 '41' '42' '43' '44' '45' '46' '47' '56' '57' '58' '9' '59' '60' '61' '62'
 '63' '10' '11' '28']
sampling frequency: 20000
Nr of channels: 64
Nr of segments: 8
Traceback (most recent call last):

  File "C:\Users\susan\AppData\Local\Temp\ipykernel_13708\204832346.py", line 16, in <module>
    rec_prb = recording.load_probe_file('32TetPrb.prb')

AttributeError: 'NeuralynxRecordingExtractor' object has no attribute 'load_probe_file'

alejoe91, 2022-08-24:
Hi @SusanL82 

As the error suggests, the `load_probe_file` function doesn't exist. In order to load the `.prb` file, you have to use `probeinterface`:

```
from probeinterface import read_prb

# different tetrodes are loaded as separate probes in the probegroup
probegroup = read_prb('32TetPrb.prb')

rec_prb = recording.set_probegroup(probegroup, group_mode='by_probe')
```

The `by_probe` mode will group different tetrodes in different `groups`. You can check with:
```
print(rec_prb.get_channel_groups())
```

SusanL82, 2022-08-30:
Thanks!
I think there is something off about my .prb (read_prb gives an error), but the channel IDs listed in there won't work for me here.  

As it turns out, se.read_neuralynx() gives my tetrode wires in the order 1, 10, 11, 12, ..., 19, 2, 20 with the recording system's channel ID to each one (e.g. CSC1.ncs is recorded on channel AD12, CSC10.ncs is AD29...). I think I'll need a more complex file to deal with that anyway.

samuelgarcia, 2022-08-30:
If you mention `load_probe_file` maybe you are reading an old presentation or tutorial.
This way of doing do not exists anymore.

The most recent tutorial is this one: 
https://github.com/SpikeInterface/spiketutorials/tree/master/Official_Tutorial_SI_0.94_May2022

SusanL82, 2022-09-07:
O, that's good to know. I didn't realize that the tutorials were so different.

I ended up using recording.set_channel_groups() and recording.set_channel_locations() directly instead. It seems to work ok now (there is an issue with mountainsort, but that seems unrelated to the channel mapping).

closed 2022-09-07 by SusanL82

samuelgarcia, 2022-09-07:
An even more recent tutorial were published 2 day ago for the 0.95 releas (3 day ago)
https://github.com/SpikeInterface/spiketutorials/tree/master/Official_Tutorial_SI_0.95_Sep2022