SpikeInterface/spikeinterface #907: Neuralynx data: .nrd and .ncs files

dejanica93, 2022-08-24:
Hello!

I was wondering how to convert the .nrd files (raw data) and also all the .ncs files from all the channels (continuous data; one file per channel) to be able to run in through spikeinterface for sorting.
Thank you!

h-mayorquin, 2022-08-30:
@dejanica93 
Did you manage to load the files, I have been trying to get some more experience with neuralynx files recently and would love to help you if you are still into this.

caniko, 2022-08-30:
@samuelgarcia, I have been helping Dejana today, and it seems like `__getitem__` is not implemented for `NeuralynxRecordingExtractor`. This is an issue because `si.concatenate_recordings` fails to when running [`si.append_recordings`](https://github.com/SpikeInterface/spikeinterface/blob/5debc19d6c69f9c4acf10dfabaeafeca62a10407/spikeinterface/core/segmentutils.py#L59) when it call index `0` on `NeuralynxRecordingExtractor` object.

I'll keep looking into this, maybe I can post a PR with a fix.

caniko, 2022-08-30:
The bug was on our side! I have a PR idea that I will post because I think the `si.concatenate_recordings` is a little unforgiving when the number of recordings is one, and it makes no sense to provide a list of recordings.

I also had trouble running kilosort3 with docker, but that is another issue.

caniko, 2022-09-01:
> @dejanica93 Did you manage to load the files, I have been trying to get some more experience with neuralynx files recently and would love to help you if you are still into this.

We managed to load the files indeed. Just use the [neuralynx reader function](https://spikeinterface.readthedocs.io/en/latest/api.html?spikeinterface.extractors.read_neuralynx#spikeinterface.extractors.read_neuralynx).

h-mayorquin, 2022-09-01:
I am not stuck, I am trying to improve some readers and therefore I want to get other examples as the only ones that I have are in the gin tests : )

Thanks for the update!

dejanica93, 2022-09-01:
Everything seems to work! 
Thank you all for the help!

closed 2022-09-01 by dejanica93