SpikeInterface/spikeinterface #913: read open ephys v0.6

shirquinn, 2022-08-25:
Hi @alejoe91 
when i try to read a binary file recorded with open ephys gui version 0.6.0 i get this error
the files are ordered differently in this new version, any chance you can adapt the code to read the files from the new version?

`ValueError                                Traceback (most recent call last)
Input In [7], in <cell line: 1>()
----> 1 recording = si.read_openephys(recording_file)

File ~\anaconda3\lib\site-packages\spikeinterface\extractors\neoextractors\openephys.py:117, in read_openephys(folder_path, **kwargs)
    114     recording = OpenEphysLegacyRecordingExtractor(folder_path, **kwargs)
    115 else:
    116     # format = 'binary'
--> 117     recording = OpenEphysBinaryRecordingExtractor(folder_path, **kwargs)
    118 return recording

File ~\anaconda3\lib\site-packages\spikeinterface\extractors\neoextractors\openephys.py:66, in OpenEphysBinaryRecordingExtractor.__init__(self, folder_path, stream_id)
     64 def __init__(self, folder_path, stream_id=None):
     65     neo_kwargs = {'dirname': folder_path}
---> 66     NeoBaseRecordingExtractor.__init__(self, stream_id=stream_id, **neo_kwargs)
     68     probe = pi.read_openephys(folder_path, raise_error=False)
     69     if probe is not None:

File ~\anaconda3\lib\site-packages\spikeinterface\extractors\neoextractors\neobaseextractor.py:26, in NeoBaseRecordingExtractor.__init__(self, stream_id, **neo_kwargs)
     24 def __init__(self, stream_id=None, **neo_kwargs):
---> 26     _NeoBaseExtractor.__init__(self, **neo_kwargs)
     28     # check channel
     29     # TODO propose a meachanisim to select the appropriate channel groups
     30     # in neo one channel group have the same dtype/sampling_rate/group_id
     31     # ~ channel_indexes_list = self.neo_reader.get_group_signal_channel_indexes()
     32     stream_channels = self.neo_reader.header['signal_streams']

File ~\anaconda3\lib\site-packages\spikeinterface\extractors\neoextractors\neobaseextractor.py:16, in _NeoBaseExtractor.__init__(self, **neo_kwargs)
     14 neoIOclass = eval('neo.rawio.' + self.NeoRawIOClass)
     15 self.neo_reader = neoIOclass(**neo_kwargs)
---> 16 self.neo_reader.parse_header()
     18 assert self.neo_reader.block_count() == 1, \
     19     'This file is neo multi block spikeinterface support one block only dataset'

File ~\AppData\Roaming\Python\Python39\site-packages\neo-0.10.2-py3.9.egg\neo\rawio\baserawio.py:185, in BaseRawIO.parse_header(self)
    172 def parse_header(self):
    173     """
    174     This must parse the file header to get all stuff for fast use later on.
    175 
   (...)
    183 
    184     """
--> 185     self._parse_header()
    186     self._check_stream_signal_channel_characteristics()

File ~\AppData\Roaming\Python\Python39\site-packages\neo-0.10.2-py3.9.egg\neo\rawio\openephysbinaryrawio.py:140, in OpenEphysBinaryRawIO._parse_header(self)
    138         d['labels'] = d['channels'].astype('U')
    139     else:
--> 140         raise ValueError(f'There is no possible labels for this event: {stream_name}')
    142 # no spike read yet
    143 # can be implemented on user demand
    144 spike_channels = np.array([], dtype=_spike_channel_dtype)

ValueError: There is no possible labels for this event: Record Node 113#Acquisition_Board-100.Rhythm Data/TTL/`

samuelgarcia, 2022-08-26:
Hi,
you need to install neo from source, this should be fix already.

samuelgarcia, 2022-08-26:
Maybe same as #905

shirquinn, 2022-08-28:
@samuelgarcia , pip install --upgrade neo didn't solve it for me.
i'm not sure on how to install it from the source. Can you explain how to do that?
Thanks any way
Shir

samuelgarcia, 2022-08-29:
Hi,
for source installation you need to do this
```
git clone https://github.com/SpikeInterface/spikeinterface.git
cd spikeinterface
python setup.py install
```

See https://github.com/SpikeInterface/spikeinterface#how-to-install-spikeinteface

shirquinn, 2022-08-29:
oh i thought you ment installing neo from the neo github.
I reinstalled spikeinterface and still get this warning: 
There is no possible labels for this event: Record Node 103#Acquisition_Board-100.Rhythm Data/TTL/

alejoe91, 2022-09-02:
@shirquinn we just released NEO, ProbeInterface, and SpikeInterface. Should be fixed if you do:

```
pip install -U neo
pip install -U probeinterface
pip install -U spikeinterface
```

alejoe91, 2022-09-05:
@shirquinn for seaborn you should be able to remove the warning by upgrading the package. For the plot function, that's also just a warning or is it triggering an error?

closed 2022-10-13 by alejoe91