SpikeInterface/spikeinterface #914: Specification of the dtype for Preprocessors

yger, 2022-08-26:
I saw a discussion concerning the dtype of the whitening pre-processors, but I just want to highlight that there is some inconsistencies between the pre-processors: some can specify an output dtype (filtering, whitening), and some can not (CAR, artefacts, ...)

Maybe there should be something more abstract and general such that all pre-processors can control the dtype output?

h-mayorquin, 2023-07-19:
Yes, so I can think on two solutions. 

Either throw a warning if the parent recording and the known necessary output does not match. Or be more demanding and make users transform the parent recording with `AstypeRecording` if we believe this is a serious error that they should not do (similar to what we do with  the `is_filtered` flag).

In any case, then, it seems that dtype should not be in any `__init__`.

Let's see if @samuelgarcia wants to chime in.

h-mayorquin, 2023-09-26:
> For some preprocessors the output will necessarily change. E.g., zscore and whitening will output a float no matter what!

This and https://github.com/SpikeInterface/spikeinterface/issues/2029 have made me change my mind. It does not makes sense that pre-processing takes the input of their parent for some cases. Now I feel that some processors like dtype and phase shift should have their own dtype and we should thrown a warning if there is an increase in memory representation.

I still believe that control of dtype should be done iwth AstypeRecording instead in the `__init__`.