SpikeInterface/spikeinterface #917: Trouble running Herdingspikes on Spikeinterface

nasnass, 2022-09-01:
Hi!
I am having a bit of difficulties getting Herdingspikes to run using Spike interface. This is the error I am getting:

The extractor was not dumpable
Error running herdingspikes
Traceback (most recent call last):
File "C:\Users\nasya\anaconda3\envs\herdingspikes\lib\site-packages\spikeinterface\sorters\basesorter.py", line 200, in run_from_folder
SorterClass._run_from_folder(output_folder, sorter_params, verbose)
File "C:\Users\nasya\anaconda3\envs\herdingspikes\lib\site-packages\spikeinterface\sorters\herdingspikes\herdingspikes.py", line 170, in _run_from_folder
recording = st.normalize_by_quantile(
File "C:\Users\nasya\anaconda3\envs\herdingspikes\lib\site-packages\spikeinterface\toolkit\preprocessing\normalize_scale.py", line 196, in normalize_by_quantile
return NormalizeByQuantileRecording(*args, **kwargs)
File "C:\Users\nasya\anaconda3\envs\herdingspikes\lib\site-packages\spikeinterface\toolkit\preprocessing\normalize_scale.py", line 60, in init
random_data = get_random_data_chunks(recording, **random_chunk_kwargs)
File "C:\Users\nasya\anaconda3\envs\herdingspikes\lib\site-packages\spikeinterface\toolkit\utils.py", line 36, in get_random_data_chunks
for segment_index in range(recording.get_num_segments()):
AttributeError: 'NoneType' object has no attribute 'get_num_segments'

Here is how I am running it:

sorting_HS = ss.run_herdingspikes(recording, output_folder='results_HS', filter=False, verbose=True)

Whenever I try to pre-process it, and then run the sorter, the Kernel dies.

recording_f = st.bandpass_filter(recording, freq_min=100, freq_max=3000)
recording = recording_f.save(format='binary')

I have only take a small chunk of my data (not even 1GB) just to try and get it to work. Would be very thankful for any hints! I have already talked to the team at Herdingspikes and they are saying it might have something to do with the pre-processing of SI.

nasnass, 2022-09-01:
@samuelgarcia I have updated it, still same issue.

alejoe91, 2022-10-06:
@nasnass how are you running HerdingSpikes? I think there might be an error in how you pass the arguments, because the `recording` seems to be None...

closed 2023-03-27 by alejoe91