SpikeInterface/spikeinterface #918: si.extract_waveforms don't compute templates if folder exists

Dradeliomecus, 2022-09-01:
Hi,

I came across a (not very likely) scenario where an unintended behaviour happens.

If you call `si.extract_waveforms(..., load_if_exists=True, precompute_template=("average", ))`, and the folder already exists **but** the file `templates_average.npy` is missing, then it will not precompute the templates.

In my case it was relevant because SpikeInterface crashed during the templates computation due to a "too many open files error".
But I guess it can also be relevant if someone adds another template mode (e.g. median) and the folder already exists.

alejoe91, 2022-12-28:
@DradeAW we're planning to remove the `load_if_exists` in favor of `si.load_waveforms()`.

Then what you would do is:
```
we = si.load_waveforms("existing-folder")
we.precompute_templates(modes=('average', )):
```

DradeAW, 2022-12-28:
Okay, seems like a good solve!

closed 2023-01-27 by alejoe91