SpikeInterface/spikeinterface #930: EDF EXTRACTOR

olesnits, 2022-09-07:
Hi,

I have used Edf Extractor with many files (MEA recordings), but always I have the same prolem - metadata of the file is not properly read (wrong time in seconds, wrong sampling frequency.) I guess the problem how it reads header. Hope you can fix it!

olesnits, 2022-09-07:
I have checked the same files with H5 extractor, and metadata was read correctly.

alejoe91, 2022-09-07:
@h-mayorquin can you take a look at this? :)

h-mayorquin, 2022-09-08:
@olesnits Do you think you could share your data with us?

olesnits, 2022-09-09:
hello! sorry, i can't. data is private from patient's. I can share with you notebook, where i open file with edf extractor and h5 extractor to see comparison in meta-data. or i can be avalaible for call to open data with you. let me know

closed 2022-10-13 by alejoe91