SpikeInterface/spikeinterface #932: ArrayMemoryError when using tridesclous

phanhuynh, 2022-09-08:
Hello team,
I've attached the error log for tridesclous.  
The key phrase is:
```
raise self._exception\nnumpy.core._exceptions._ArrayMemoryError: Unable to allocate 10.2 GiB for an array with shape (42923008, 64) and data type float32\n",
    "run_time": null
```
Any thoughts?  Thank you very much!
[spikeinterface_log.txt](https://github.com/SpikeInterface/spikeinterface/files/9526041/spikeinterface_log.txt)

caniko, 2022-09-08:
Seems like you are out of memory (OOM), how much RAM do you have? You can also [increase your swap memory](https://superuser.com/questions/793304/how-to-increase-swap-memory-in-windows), but this will impact performance.

Unless there is a setting for PCs with low RAM, the best thing to do is to get a PC with more RAM.

phanhuynh, 2022-09-13:
Hello Can, I'm not sure if my response came through via email (8 Sept):
Hi Can, 
Thank you for your response.  
However, the RAM on this computer is 128Go 3200 MHz DDR4 - 4x32Go ECC.
Seems like plenty.  What do you think?

alejoe91, 2022-09-13:
@samuelgarcia thoughts?

caniko, 2022-09-13:
I don't know what kind of raw data you have, and why the array is so big, do you have 64 channels? I would use my NVME drive as swap or re-design my datastructures. Swap is really easy because it doesn't require much effort, and it can tell you how much memory the sorter really needs.

Did you try any other sorters?

phanhuynh, 2022-09-13:
Hi Can, thank you for the quick response. 
Yes, this same data set worked with Kilosort3, Spyking Circus, and Iron Cluster.  
I'll check if the ram is maxed out when I run it.  I'll also loked into Swap.  Thanks!

phanhuynh, 2022-09-27:
Hello team!  
I've observed the RAM.  Nothing special happens but then it will max out to 100% right before crashing. 
I've tried a few things from this [stackoverflow page](https://stackoverflow.com/questions/57507832/unable-to-allocate-array-with-shape-and-data-type):
1) check that the python shell is 64bits instead of 32bits (it was correctly 64bits).
2) I've changed the overcommit handling, and changed the virtual memory limits.  This didn't help (recall that I have 128GB Ram).

Now I'm considering changing the dtype attribute from float64 to uint8, but not sure if that's a paramater that I can change.
  
Any advice?  Thank you!

caniko, 2022-09-27:
I would advise against fp64 to uint8. You could try fp32 or even fp16, wouldn't go lower.

Note that your data will lose quality when deviating from the original to lower precision.

I would rather use a lot of swap. How much swap did you set?

caniko, 2022-09-29:
Tridesclous, and other spikesorters do have minimum requirements; however, the requirements are also affected by your own data. As I mentioned earlier, your data seems to be quite different from the usual. The data has the downstream effect of requiring larger `np.ndarray`s, which crashes the runtime because of system out of memory or OOM. 

Also storage, 2TB, has no impact on this issue. This is strictly memory, which should be our focus.

Debugging points:
- How large is your raw data?
- Do you see the virtual memory in task manager? Total memory in task manager should be 384Gb after setting SWAP. Your description seems like you have been succesfull.
- Actually I am not certain you actually do experience OOM when setting your overall memroy to 384Gb. Did you see the entirity of 384Gb being filled up? I would start leaning towards the possibility of a bug if this is the case, because this is a lot of memory use.

samuelgarcia, 2022-09-29:
Hi,
I am sorry for so long delay.
Normally tridesclous is doing chunk processing. So normally it cusums very few ram (depending chunk size of course).

Norammly the input is memmaped (using np.memmap) so normally we can adress very very dataset with a few ram.
The OS should load onde mand every chunk in memory and clean it when not necessary anymore.

Normally tridesclous should not triggered swap at all. I don't have swap on my own machine, I hate that I prefer true memory errore!

There must be a bug somewhere if tridesclous consum too much ram.
Could you running tridesclous with the docker ?

Unfortunatly tridesclous is very weel maintain since 2 years because my effort are now in spikeinterface.
I am working now on tridesclous2  which will be spikeinterface base and normaly efficient in computation time and memory consumption.

Sorry not to able to help more.

phanhuynh, 2022-09-30:
Hi Sam - no worries, I know you're super busy.  Great idea though about using docker.  Unfortunately, I get the error pasted below.
@Can, thank you for all your ideas!  I'll try them if I can't get docker to work.  But to answer your first question, the raw file is 20.6GB.

```
Starting container
Installing spikeinterface==0.95.1 in spikeinterface/tridesclous-base
Installing extra requirements: ['neo']
Running tridesclous sorter inside spikeinterface/tridesclous-base
Stopping container
---------------------------------------------------------------------------
SpikeSortingError                         Traceback (most recent call last)
Input In [9], in <cell line: 17>()
      1 #####################################################
      2 ### Option 1: Run local sorter (recommended) ###########
      3 #####################################################
   (...)
     14 ### Option 2: run sorter directly from docker ###
     15 #####################################################
     16 start_time = time.time()
---> 17 sorting_TDC = ss.run_sorter(
     18     'tridesclous',
     19     recording_saved,
     20     output_folder= '_output_TDC',
     21     remove_existing_folder=True,
     22     detect_threshold=threshold,
     23     docker_image="spikeinterface/tridesclous-base", 
     24     verbose=True,
     25 )
     26 print("--- _output_TDC took %s seconds ---" % (time.time() - start_time))
     27 print(f'Tridesclous found {len(sorting_TDC.get_unit_ids())} units')

File ~\Anaconda3\envs\sienv\lib\site-packages\spikeinterface\sorters\runsorter.py:136, in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, singularity_image, with_output, **sorter_params)
    134         else:
    135             container_image = singularity_image
--> 136     return run_sorter_container(
    137         container_image=container_image,
    138         mode=mode,
    139         **common_kwargs,
    140     )
    142 return run_sorter_local(**common_kwargs)

File ~\Anaconda3\envs\sienv\lib\site-packages\spikeinterface\sorters\runsorter.py:548, in run_sorter_container(sorter_name, recording, mode, container_image, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, extra_requirements, **sorter_params)
    546 if run_error:
    547     if raise_error:
--> 548         raise SpikeSortingError(
    549             f"Spike sorting in {mode} failed with the following error:\n{run_sorter_output}")
    550 else:
    551     if with_output:

SpikeSortingError: Spike sorting in docker failed with the following error:
b'Traceback (most recent call last):\n  File "/Users/huynh/spike_epsztein/Phan/proj_nexus_v2/in_container_sorter_script.py", line 17, in <module>\n    sorting = run_sorter_local(\n  File "/usr/local/lib/python3.8/site-packages/spikeinterface/sorters/runsorter.py", line 157, in run_sorter_local\n    output_folder = SorterClass.initialize_folder(\n  File "/usr/local/lib/python3.8/site-packages/spikeinterface/sorters/basesorter.py", line 123, in initialize_folder\n    shutil.rmtree(str(output_folder))\n  File "/usr/local/lib/python3.8/shutil.py", line 718, in rmtree\n    _rmtree_safe_fd(fd, path, onerror)\n  File "/usr/local/lib/python3.8/shutil.py", line 655, in _rmtree_safe_fd\n    _rmtree_safe_fd(dirfd, fullname, onerror)\n  File "/usr/local/lib/python3.8/shutil.py", line 655, in _rmtree_safe_fd\n    _rmtree_safe_fd(dirfd, fullname, onerror)\n  File "/usr/local/lib/python3.8/shutil.py", line 675, in _rmtree_safe_fd\n    onerror(os.unlink, fullname, sys.exc_info())\n  File "/usr/local/lib/python3.8/shutil.py", line 673, in _rmtree_safe_fd\n    os.unlink(entry.name, dir_fd=topfd)\nPermissionError: [Errno 13] Permission denied: \'all_peaks.raw\'\n'
```

caniko, 2022-09-30:
Running on docker and failing would eliminate the possibility pf wrongful installation.

You can also try my repacked version pf spikeinterface with trideclous `pip install spikeinterface-poetry[tridesclous]` (this is supported only by myself). I didn't mention it because it seems like you are running the sorter, low chance of wrongful install.

Wish you the best of luck!

caniko, 2022-10-11:
Yes, it is plenty; however, all it takes for OOM is either bugged code and/or loading too much into memory. Hope this was made clear earlier.

phanhuynh, 2022-10-27:
Hello team, 
The problem went away when I updated Spikeinterface ( `!pip install --upgrade spikeinterface`).
I can't believe I didn't try that earlier -- gaaaaaah!
Thank you again for your support!

alejoe91, 2022-10-27:
Yaaaaayy!!! What version were you using?

alejoe91, 2022-10-27:
Not sure, but glad it is solved!

closed 2022-10-27 by alejoe91