SpikeInterface/spikeinterface #933: `run_sorters()` with docker should run fine on Windows

caniko, 2022-09-08:
Hello, we were trying to get the spike sorters up with docker and running on the personal PCs on the lab today. I thought this would be OK considering that WSL 2 makes GPU mounting possible on Windows.

To my surprise this is not supported on the `spikeinterface` side, we got an `AssertionError`. What exactly is the issue on the Windows side? @alejoe91 @samuelgarcia 

Kind regards

alejoe91, 2022-09-08:
Can you paste the error?

dejanica93, 2022-09-08:
Input In [14], in <cell line: 3>()
      1 import spikeinterface.sorters as ss
----> 3 sorting = ss.run_kilosort3(
      4     recording=recording,
      5     output_folder="kilosort3",
      6     singularity_image=True)
      8 print(sorting)

File ~\PycharmProjects\spikeinterface\venv\lib\site-packages\spikeinterface\sorters\runsorter.py:454, in run_kilosort3(*args, **kwargs)
    453 def run_kilosort3(*args, **kwargs):
--> 454     return run_sorter('kilosort3', *args, **kwargs)

File ~\PycharmProjects\spikeinterface\venv\lib\site-packages\spikeinterface\sorters\runsorter.py:62, in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, singularity_image, with_output, **sorter_params)
     56     return run_sorter_container(sorter_name, recording, 'docker', docker_image,
     57                                 output_folder=output_folder,
     58                                 remove_existing_folder=remove_existing_folder,
     59                                 delete_output_folder=delete_output_folder, verbose=verbose,
     60                                 raise_error=raise_error, with_output=with_output, **sorter_params)
     61 if singularity_image is not None:
---> 62     return run_sorter_container(sorter_name, recording, 'singularity', singularity_image,
     63                                 output_folder=output_folder,
     64                                 remove_existing_folder=remove_existing_folder,
     65                                 delete_output_folder=delete_output_folder, verbose=verbose,
     66                                 raise_error=raise_error, with_output=with_output, **sorter_params)
     67 return run_sorter_local(sorter_name, recording, output_folder=output_folder,
     68                         remove_existing_folder=remove_existing_folder,
     69                         delete_output_folder=delete_output_folder,
     70                         verbose=verbose, raise_error=raise_error, with_output=with_output,
     71                         **sorter_params)

File ~\PycharmProjects\spikeinterface\venv\lib\site-packages\spikeinterface\sorters\runsorter.py:222, in run_sorter_container(sorter_name, recording, mode, container_image, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, **sorter_params)
    217 def run_sorter_container(sorter_name, recording, mode, container_image, output_folder=None,
    218                          remove_existing_folder=True, delete_output_folder=False,
    219                          verbose=False, raise_error=True, with_output=True, **sorter_params):
    220     # common code for docker and singularity
--> 222     assert platform.system() in ('Linux', 'Darwin'), \
    223         'run_sorter() with docker is supported only on linux/macos platform '
    225     if output_folder is None:
    226         output_folder = sorter_name + '_output'

AssertionError: run_sorter() with docker is supported only on linux/macos platform

alejoe91, 2022-09-08:
Can you updgrade to 0.95.1? Windows is now supported

alejoe91, 2022-09-22:
@caniko @dejanica93 did you try with new version? 

If it works, I'm going to close the issue!

dejanica93, 2022-09-28:
Yes, the new version worked, but now I have issues with running kilosort.
Sorry for the late response, and thanks for help!

closed 2022-09-28 by alejoe91