SpikeInterface/spikeinterface #974: Would be helpful to have tutorial for merged sorting

BenATemple, 2022-09-28:
It might be helpful to put a tutorial for MergeUnitsSorting on the Read The Docs.
I can merge one set of units at a time, but not more (so can't do my whole set of units).
This would be really helpful, unless there is a way to re-import merging done in Phy.
Thanks for your help!

alejoe91, 2022-09-28:
@BenATemple of course you can reimport merging done in Phy!

```
sorting_curated = se.read_phy("path-to-phy-folder", exclude_cluster_groups=["noise"])
```

Anyways, the `curation` module is still under development but we'll add some docs soon!

BenATemple, 2022-09-28:
Very nice, thank you!

closed 2022-10-13 by alejoe91

rory-bedford, 2023-12-15:
Is there a way to load a waveform extractor object directly from the phy folder? I feel like all the necessary data should be there, but can't figure out how to do it. At the moment I'm having to load the sorting object from phy as above, and load my original recording from elsewhere, then extract the waveforms again, which might be unnecessary but I'm not sure.