SpikeInterface/spikeinterface #978: read_intan returns a recording with 0 duration

hpay, 2022-09-29:
I'm reading intan data (single file per signal type) as follows:
```
# Load using intan extractor
data_file = Path('Z:/Hannah/ephys/project2/test_rec/raw/info.rhd')
recording_raw = si.read_intan(data_file, stream_id='0')

channel_ids = recording_raw.get_channel_ids()
fs = recording_raw.get_sampling_frequency()
num_chan = recording_raw.get_num_channels()
num_segments = recording_raw.get_num_segments()
total_duration = recording_raw.get_total_duration()

print(recording_raw)
print(f'Channel ids: {channel_ids}')
print(f'Sampling frequency: {fs}')
print(f'Number of channels: {num_chan}')
print(f"Number of segments: {num_segments}")
print(f"Total duration: {total_duration}")
```

Everything is correct except total_duration  = 0s, which of course causes any further operations to fail. 

When I read it the same file as binary it is ok (Total duration: 206.6 s):
```
recording_file = Path('Z:/Hannah/ephys/project2/test_rec/raw/amplifier.dat')
print(recording_file)
num_channels = 64
sampling_frequency = 30000
gain_to_uV = 0.195
offset_to_uV = 0
dtype = "int16"
time_axis = 1
recording_raw = si.read_binary(recording_file, num_chan=num_channels, sampling_frequency=sampling_frequency,
                           dtype=dtype, gain_to_uV=gain_to_uV, offset_to_uV=offset_to_uV, 
                           time_axis=time_axis)
```

I'd be happy to share this sample file in case that would help. It would be nice to use the read_intan function to avoid needing to set all the parameters manually. Thank you in advance.

samuelgarcia, 2022-09-29:
Hi.
the read_intan is based on neo implemenattion here https://github.com/NeuralEnsemble/python-neo/blob/master/neo/rawio/intanrawio.py.

This read_intan is really for the old formats : rhs and rhd.

I think the binary intan is somehow more recent and is not handle in read_intan.

hpay, 2022-09-29:
Got it, thanks!

In case it's helpful for anyone else I wrote a small wrapper module to do this:

```
# Module for importing Intan data saved in "One File Per Signal Type" format
def read_intan_dat(data_path):

    import spikeinterface.extractors as se
    import spikeinterface.core as sc

    info_file = data_path/'info.rhd'
    rec_info = se.read_intan(info_file, stream_id='0')

    sampling_frequency = rec_info.get_sampling_frequency()
    num_channels = rec_info.get_num_channels()
    num_segments = rec_info.get_num_segments() # Should be 1 - could add test?

    # Loading the .dat as a generic binary
    recording_file  = data_path/'amplifier.dat'
    gain_to_uV = 0.195
    offset_to_uV = 0
    dtype = "int16"
    time_axis = 0
    recording = sc.read_binary(recording_file, num_chan=num_channels, sampling_frequency=sampling_frequency,
                            dtype=dtype, gain_to_uV=gain_to_uV, offset_to_uV=offset_to_uV, 
                            time_axis=time_axis)
    return recording
```

closed 2022-09-29 by hpay