SpikeInterface/spikeinterface #994: All the other sorter failed except of tridesclous

betty6you, 2022-10-10:
Hi 

I am using SpikeInterface to perform spike sorting on my binary file extracted from EDF.

But only "tridesclous" worked so far, the other sorters all failed. 

Here is a copy of my code and output. https://github.com/betty6you/EDFRecordingExtractor/blob/main/EDF_extractor_test_1.ipynb

What will be the reason and how to solve it?

Many thanks!
Jing

alejoe91, 2022-10-13:
HI @betty6you 

Looking at the notebook it seems that tridesclous only found 1 good unit..Is it possible that the recording doesn't have any clearly isolated unit? This could explain the spike sorters failure!

betty6you, 2022-10-13:
Thanks for your explanation.
I will try another file today.

closed 2022-12-28 by alejoe91