SpikeInterface/spikeinterface #1039: Error with pykilosort

PicniCat, 2022-10-28:
Hi,

I had a problem when running pykilosort. It's an index error: `IndexError: Index 0 is out of bounds for axis 1 with size 0`. Here's the code and results:
Running:
```
sorting = ss.run_sorter('pykilosort', recording_rm_artifact)
```
Returns:
```
Computing the whitening matrix: 100%
1/1 [00:00<00:00, 3.16it/s]
Preprocessing: 100%
21/21 [00:01<00:00, 20.01it/s]
Extracting templates: 100%
5/5 [00:00<00:00, 138.24it/s]
Traceback (most recent call last):
  File "C:\Anaconda3\lib\site-packages\spikeinterface\sorters\basesorter.py", line 225, in run_from_folder
    SorterClass._run_from_folder(output_folder, sorter_params, verbose)
  File "C:\Anaconda3\lib\site-packages\spikeinterface\sorters\pykilosort\pykilosort.py", line 180, in _run_from_folder
    run(
  File "C:\Anaconda3\lib\site-packages\pykilosort-ibl_1.3.3-py3.9.egg\pykilosort\main.py", line 200, in run
    out = learnAndSolve8b(ctx)
  File "C:\Anaconda3\lib\site-packages\pykilosort-ibl_1.3.3-py3.9.egg\pykilosort\learn.py", line 847, in learnAndSolve8b
    wTEMP, wPCA = extractTemplatesfromSnippets(
  File "C:\Anaconda3\lib\site-packages\pykilosort-ibl_1.3.3-py3.9.egg\pykilosort\learn.py", line 69, in extractTemplatesfromSnippets
    wTEMP[:, j] = cp.dot(dd[:, imax == j], amax[imax == j].T)
  File "cupy/core/core.pyx", line 1299, in cupy.core.core.ndarray.__setitem__
  File "cupy/core/_routines_indexing.pyx", line 52, in cupy.core._routines_indexing._ndarray_setitem
  File "cupy/core/_routines_indexing.pyx", line 881, in cupy.core._routines_indexing._scatter_op
  File "cupy/core/_routines_indexing.pyx", line 401, in cupy.core._routines_indexing._simple_getitem
IndexError: Index 0 is out of bounds for axis 1 with size 0

---------------------------------------------------------------------------
SpikeSortingError                         Traceback (most recent call last)
~\AppData\Local\Temp/ipykernel_16060/592903683.py in <module>
      1 print(ss.installed_sorters())
----> 2 sorting = ss.run_sorter('pykilosort', recording_rm_artifact)

C:\Anaconda3\lib\site-packages\spikeinterface\sorters\runsorter.py in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, singularity_image, with_output, **sorter_params)
    141         )
    142 
--> 143     return run_sorter_local(**common_kwargs)
    144 
    145 

C:\Anaconda3\lib\site-packages\spikeinterface\sorters\runsorter.py in run_sorter_local(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, **sorter_params)
    161         recording, output_folder, sorter_params, verbose)
    162     SorterClass.setup_recording(recording, output_folder, verbose=verbose)
--> 163     SorterClass.run_from_folder(output_folder, raise_error, verbose)
    164     if with_output:
    165         sorting = SorterClass.get_result_from_folder(output_folder)

C:\Anaconda3\lib\site-packages\spikeinterface\sorters\basesorter.py in run_from_folder(cls, output_folder, raise_error, verbose)
    259         if has_error and raise_error:
    260             print(log['error_trace'])
--> 261             raise SpikeSortingError(
    262                 f"Spike sorting failed. You can inspect the runtime trace in {output_folder}/spikeinterface_log.json")
    263 

SpikeSortingError: Spike sorting failed. You can inspect the runtime trace in pykilosort_output/spikeinterface_log.json
```
Could you take a look and provide me with some advice? 
The content of log.json is attached below:
```
{
    "sorter_name": "pykilosort",
    "sorter_version": "ibl_1.3.3",
    "datetime": "2022-10-28T19:04:38.722232",
    "runtime_trace": [],
    "error": true,
    "error_trace": "Traceback (most recent call last):\n  File \"C:\\Anaconda3\\lib\\site-packages\\spikeinterface\\sorters\\basesorter.py\", line 225, in run_from_folder\n    SorterClass._run_from_folder(output_folder, sorter_params, verbose)\n  File \"C:\\Anaconda3\\lib\\site-packages\\spikeinterface\\sorters\\pykilosort\\pykilosort.py\", line 180, in _run_from_folder\n    run(\n  File \"C:\\Anaconda3\\lib\\site-packages\\pykilosort-ibl_1.3.3-py3.9.egg\\pykilosort\\main.py\", line 200, in run\n    out = learnAndSolve8b(ctx)\n  File \"C:\\Anaconda3\\lib\\site-packages\\pykilosort-ibl_1.3.3-py3.9.egg\\pykilosort\\learn.py\", line 847, in learnAndSolve8b\n    wTEMP, wPCA = extractTemplatesfromSnippets(\n  File \"C:\\Anaconda3\\lib\\site-packages\\pykilosort-ibl_1.3.3-py3.9.egg\\pykilosort\\learn.py\", line 69, in extractTemplatesfromSnippets\n    wTEMP[:, j] = cp.dot(dd[:, imax == j], amax[imax == j].T)\n  File \"cupy/core/core.pyx\", line 1299, in cupy.core.core.ndarray.__setitem__\n  File \"cupy/core/_routines_indexing.pyx\", line 52, in cupy.core._routines_indexing._ndarray_setitem\n  File \"cupy/core/_routines_indexing.pyx\", line 881, in cupy.core._routines_indexing._scatter_op\n  File \"cupy/core/_routines_indexing.pyx\", line 401, in cupy.core._routines_indexing._simple_getitem\nIndexError: Index 0 is out of bounds for axis 1 with size 0\n",
    "run_time": null
}
```

Many thanks!

alejoe91, 2022-11-03:
Hi, what version of pykilosort are you using? Can you try using the latest version from IBL? https://github.com/int-brain-lab/pykilosort

alejoe91, 2022-12-28:
@PicniCat any update on this?

yinrocky, 2023-01-01:
@alejoe91  I am using the latest version from IBL(pykilosort-ibl_1.4.0). and have the same issue
{
    "sorter_name": "pykilosort",
    "sorter_version": "ibl_1.4.0",
    "datetime": "2022-12-31T20:06:59.844094",
    "runtime_trace": [],
    "error": true,
    "error_trace": "Traceback (most recent call last):\n  File \"/home/rongkang/anaconda3/envs/si_env/lib/python3.9/site-packages/spikeinterface/sorters/basesorter.py\", line 225, in run_from_folder\n    SorterClass._run_from_folder(output_folder, sorter_params, verbose)\n  File \"/home/rongkang/anaconda3/envs/si_env/lib/python3.9/site-packages/spikeinterface/sorters/pykilosort/pykilosort.py\", line 180, in _run_from_folder\n    run(\n  File \"/home/rongkang/anaconda3/envs/si_env/lib/python3.9/site-packages/pykilosort-ibl_1.4.0-py3.9.egg/pykilosort/main.py\", line 177, in run\n    out = learnAndSolve8b(ctx)\n  File \"/home/rongkang/anaconda3/envs/si_env/lib/python3.9/site-packages/pykilosort-ibl_1.4.0-py3.9.egg/pykilosort/learn.py\", line 847, in learnAndSolve8b\n    wTEMP, wPCA = extractTemplatesfromSnippets(\n  File \"/home/rongkang/anaconda3/envs/si_env/lib/python3.9/site-packages/pykilosort-ibl_1.4.0-py3.9.egg/pykilosort/learn.py\", line 69, in extractTemplatesfromSnippets\n    wTEMP[:, j] = cp.dot(dd[:, imax == j], amax[imax == j].T)\n  File \"cupy/_core/core.pyx\", line 1590, in cupy._core.core._ndarray_base.__setitem__\n  File \"cupy/_core/_routines_indexing.pyx\", line 52, in cupy._core._routines_indexing._ndarray_setitem\n  File \"cupy/_core/_routines_indexing.pyx\", line 936, in cupy._core._routines_indexing._scatter_op\n  File \"cupy/_core/_routines_indexing.pyx\", line 420, in cupy._core._routines_indexing._view_getitem\nIndexError: Index 0 is out of bounds for axis 1 with size 0\n",
    "run_time": null
}

alejoe91, 2023-01-01:
Does this happen for all recordings?

yinrocky, 2023-01-01:
> Does this happen for all recordings?

yes, tried two datasets(128channels) recorded with Intan and converted to NWB files.

yinrocky, 2023-01-02:
An update on this issue, it seems the problem comes from computing the whitening matrix. params.normalisation ==[ 'whitening' ]  always returns false, so epsilon is 1e-6 all the time. It is resolved by changing the params.normalisation ==[ 'whitening' ]  to  params.normalisation =='whitening' in the following code:
/home/rongkang/anaconda3/envs/si_env/lib/python3.9/site-packages/pykilosort-ibl_1.4.0-py3.9.egg/pykilosort/preprocess.py
line 252 
epsilon = np.mean(np.diag(CC)[probe.good_channels]) * 1e-3 if params.normalisation ==[ 'whitening' ] else 1e-6

Hope this will be helpful if someone has the same issue.

alejoe91, 2023-01-02:
Thanks for digging into this! Indeed the whitening step is problematic for some recordings! @oliche maybe worth adding this fix on the pykilosort side?

oliche, 2023-01-03:
Yes I've fixed it on 1.4.1, this is indeed not expected behaviour, thanks for the report !

However, when normalisation starts to matter, there may be something fishy in the preprocessing. 
I would still have a look at the raw data before and after pre-processing. In our case, we really targeted the destriping on neuropixel hardware, and I have no experience of other systems as the one mentioned here.

Some recipes here to look at the data before after processing:
[/datadisk/Data/neuropixel/spike_sorting/integration_test](https://int-brain-lab.github.io/iblenv/notebooks_external/loading_raw_ephys_data.html#Example-1:-Destripe-AP-data)

`pip install viewephys`
`pip install ibl-neuropixel`

yinrocky, 2023-01-04:
> Yes I've fixed it on 1.4.1, this is indeed not expected behaviour, thanks for the report !
>
> However, when normalisation starts to matter, there may be something fishy in the preprocessing. I would [...]

Thank you!

closed 2023-01-27 by alejoe91