SpikeInterface/spikeinterface #1043: [implementation] Finalize “destriping” IBL preprocessing

alejoe91, 2022-10-31:
The IBL's preprocessing pipeline includes the following steps which need to be ported and tested in SpikeInterface.
Reference white paper here: https://figshare.com/articles/online_resource/Spike_sorting_pipeline_for_the_International_Brain_Laboratory/19705522/3
 
- [ ] Bad channel detection
- [ ] Bad channel interpolation/removal 
- [x] Phase shift (implemented [here]())
- [ ] Spatial `kfilt` filter 

### Additional comments

[JoeZiminski](https://github.com/JoeZiminski) from Cambridge University is working on this: https://github.com/SpikeInterface/spikeinterface/pull/1025

### ETA since in-progress

2-3 weeks

JoeZiminski, 2022-11-04:
Thanks Alessio, bad channel interpolation and spatial filter are nearly done, just need to check a couple of things (aiming for PR today or Monday)

alejoe91, 2022-11-04:
Sounds great! No rush :)

closed 2023-03-27 by alejoe91