SpikeInterface/spikeinterface #1047: Export to phy problems

farazmoradi, 2022-11-01:
Hi,

I have done spike sorting using Kilosort 3 and export_to_phy to do manual curation. I did not receive an error while performing; however, when I wanted to use the Phy for manual curation, I received this error. It seems there is a problem with generating files by export_to_phy.

(base) E:\Spike_sorting\T9\Oct 19\phy>phy template-gui params.py --clear-state
10:30:59.954 [E] __init__:62          An error has occurred (AssertionError):
Traceback (most recent call last):
  File "C:\Users\fmora103\Anaconda3\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\fmora103\Anaconda3\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\fmora103\Anaconda3\Scripts\phy.exe\__main__.py", line 7, in <module>
    sys.exit(phycli())
  File "C:\Users\fmora103\Anaconda3\lib\site-packages\click\core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\fmora103\Anaconda3\lib\site-packages\click\core.py", line 1053, in main
    rv = self.invoke(ctx)
  File "C:\Users\fmora103\Anaconda3\lib\site-packages\click\core.py", line 1659, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\fmora103\Anaconda3\lib\site-packages\click\core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\fmora103\Anaconda3\lib\site-packages\click\core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "C:\Users\fmora103\Anaconda3\lib\site-packages\click\decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "C:\Users\fmora103\Anaconda3\lib\site-packages\phy\apps\__init__.py", line 159, in cli_template_gui
    template_gui(params_path, **kwargs)
  File "C:\Users\fmora103\Anaconda3\lib\site-packages\phy\apps\template\gui.py", line 209, in template_gui
    model = load_model(params_path)
  File "C:\Users\fmora103\Anaconda3\lib\site-packages\phylib\io\model.py", line 1348, in load_model
    return TemplateModel(**get_template_params(params_path))
  File "C:\Users\fmora103\Anaconda3\lib\site-packages\phylib\io\model.py", line 339, in __init__
    self._load_data()
  File "C:\Users\fmora103\Anaconda3\lib\site-packages\phylib\io\model.py", line 358, in _load_data
    assert self.amplitudes.shape == (ns,)
AssertionError

QWidget: Must construct a QApplication before a QWidget

alejoe91, 2022-11-01:
Hi @farazmoradi 

Can you share the phy folder?

farazmoradi, 2022-11-01:
Sure, I just started uploading!

alejoe91, 2022-11-03:
@farazmoradi I can't access the files (I need a uottawa account). Can you share them somewhere else, like on gdrive?

a-zmz, 2022-12-15:
I just had the same error, it only occurred in one of my sortings, the rest were all good. Could not work out why...

alejoe91, 2022-12-28:
I think this is related to #997 and happens when somehow KS has spikes beyond the duration of the recording...which is very weird!

a-zmz, 2023-01-09:
I think the problem was in the `spike_time.npy` file in the exported folder. For some reason, I got two spike timestamps with smaller values than their previous one (see attached plot of `t(n+1) - t(n)`, two negative values at t = 991579, 1022697).

This problem does not exists in kilosort 3 output without using `export_to_phy`.

[image]

closed 2023-03-27 by alejoe91