SpikeInterface/spikeinterface #1056: extractor.read_spikeglx results in ValueError

kushalbakshi, 2022-11-03:
I'm attempting to read a raw ephys data file acquired using `neuropixels 1.0` probes and `spikeGLX` to export it as a NWB file and I'm running into the following ValueError when using the `extractors.read_spikeglx()` function: 
```
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
Cell In [42], line 1
----> 1 extractor = extractors.read_spikeglx(npx_dir)

File c:\Users\kusha\miniconda3\envs\mapEphysNWB\lib\site-packages\spikeinterface\core\core_tools.py:27, in define_function_from_class.<locals>.reader_func(*args, **kwargs)
     25 @copy_signature(source_class)
     26 def reader_func(*args, **kwargs):
---> 27     return source_class(*args, **kwargs)

File c:\Users\kusha\miniconda3\envs\mapEphysNWB\lib\site-packages\spikeinterface\extractors\neoextractors\spikeglx.py:70, in SpikeGLXRecordingExtractor.__init__(self, folder_path, load_sync_channel, stream_id, stream_name, all_annotations)
     67 probe = pi.read_spikeglx(meta_filename)
     69 if probe.shank_ids is not None:
---> 70     self.set_probe(probe, in_place=True, group_mode="by_shank")
     71 else:
     72     self.set_probe(probe, in_place=True)

File c:\Users\kusha\miniconda3\envs\mapEphysNWB\lib\site-packages\spikeinterface\core\baserecordingsnippets.py:74, in BaseRecordingSnippets.set_probe(self, probe, group_mode, in_place)
     72 probegroup = ProbeGroup()
     73 probegroup.add_probe(probe)
---> 74 return self.set_probes(probegroup, group_mode=group_mode, in_place=in_place)

File c:\Users\kusha\miniconda3\envs\mapEphysNWB\lib\site-packages\spikeinterface\core\baserecordingsnippets.py:141, in BaseRecordingSnippets.set_probes(self, probe_or_probegroup, group_mode, in_place)
    139 # check
    140 if np.max(list(inds) + [0]) >= self.get_num_channels():
...
    142         'The given Probe have "device_channel_indices" that do not match channel count')
    143 new_channel_ids = self.get_channel_ids()[inds]
    144 arr = arr[order]

ValueError: The given Probe have "device_channel_indices" that do not match channel count
```

The export was working without this error for the same files ~ Sept. 15, 2022. Please let me know if you have any suggestions to solve this or further questions about my code and set up.

samuelgarcia, 2022-11-09:
Hi Kushal.
This is strange. Normaly, for spikeglx, the probe is handle automatically with channel indices.
Can you:
  1. try to update to latest probeinterface (we fix a bug about this recently)
  2. send us the ".meta" file from your spikeglx folder ?

alejoe91, 2022-11-22:
Hi @kushalbakshi 

Thanks for the file! The probe gets loaded correctly by `probeinterface`:

```
import probeinterface as pi
from probeinterface import plotting

probe_meta = pi.read_spikeglx("test_data.meta")
plotting.plot_probe(probe_meta)
```

Produces:
[image]

Do you have a sync channel in the `ap.bin` file? That could explain the mismatch, since the probe has 384 channels, but the bin file has 385 (384 + SYNC). If that's the case, you should load the SpikeGLX data as follows:

```
import spikeinterface.extractors as se

recording = se.read_spikeglx("path-to-folder", load_sync_channel=False)
```

Let us know if that fixes the issue!

kushalbakshi, 2022-11-22:
I actually set `load_sync_channel=True` and it worked! So I think there was a mismatch between the `ap.bin` and `.meta` files. Thank you so much for your help.

alejoe91, 2022-11-23:
Great! Closing then :)

closed 2022-11-23 by alejoe91