SpikeInterface/spikeinterface #1060: Subselecting channels

hpay, 2022-11-08:
I'm trying to exclude bad channels but the channel_slice command doesn't act as I expect.

The probe associated with my original recording looks like this:
`pi.plotting.plot_probe(probe,with_channel_index=True)
`
[image]

Just as an example, say I want to keep the first 32 channels and discard the last 32:
```
recording_raw = recording_raw.channel_slice(range(32))
probe2 = recording_raw.get_probe()
pi.plotting.plot_probe(probe2,with_channel_index=True)
```

The result is now:
[image]

I expected it to keep all the channels on the left shank, i.e. the ones labelled 0:31. Is this expected behavior?

samuelgarcia, 2022-11-08:
Hi,
`plot_probe(probe,with_channel_index=True)` is useless and totally ambiguous (I should removed it)

You should do:
  * `probeinterface.plot_probe(probe,with_contact_id=True)` : the id of the contact given by the probe manufacturer
  * `probeinterface.plot_probe(probe,with_device_index=True)` : the index (0 based) in the recording

And the best is to use the widget submodule of spikeinterface:
  * `spikeinterface.widgets.plot_probe_map(recording, with_channel_ids=True)`
because it plot directly the `channel_id`.

Importantly `rec2 = rec.channel_slice(some_channel_ids)` take **ids** and not **index**.
This looks similar when the **id** also an integer zero based but when the id is "str" or "int one base", it make a hudge diffrence.

hpay, 2022-11-08:
When I try with **with_contact_id**:
```
import probeinterface as pi
pi.plotting.plot_probe(probe,with_contact_id=True) 
```
I don't see any channels labelled.
[image]

When I try with SI's function using **with_channel_ids** (before slicing) (note I had to get rid of "widgets" for this to run) (also note, this comes out the same if I do "**with_channel_index**" instead):
`si.plot_probe_map(recording_raw, with_channel_ids=True) 
`
[image]

I see the same numbering scheme if I do "**with_device_index**" here instead. Not super helpful since I'd like to remove channels that are labelled as in the first screenshot I shared in my first post
[image]

If I do "**with_contact_id**":
[image]

Note: When I do `print(recording_raw.channel_ids)` I get:
```
[ 0  1  2  3  4  5  6  7  8  9 10 11 12 13 14 15 16 17 18 19 20 21 22 23
 24 25 26 27 28 29 30 31 32 33 34 35 36 37 38 39 40 41 42 43 44 45 46 47
 48 49 50 51 52 53 54 55 56 57 58 59 60 61 62 63]
```

I

samuelgarcia, 2022-11-08:
As you see with `si.plot_probe_map(recording_raw, with_channel_ids=True)` channel 0-31 are not on the left but sprend in the 2 shanks.
This is why when you do `recording_raw = recording_raw.channel_slice(range(32))` you get the on the two side.
Keep in mind that `channel_slice` use **id**.
You should either : gie the list of ids (which is tedious) or use the "group" property to make the split.
Have a look to that tutorial https://github.com/SpikeInterface/spiketutorials/tree/master/Official_Tutorial_SI_0.95_Sep2022
to see how the grouping is done in spikeinterface (shank or probe).

closed 2022-12-28 by alejoe91