SpikeInterface/spikeinterface #1065: spikeinterface.extract_waveforms() raises an IndexError

Yizhar-Lab, 2022-11-09:
the param's passed are:

`spikeinterface.extract_waveforms(recording, sorting, wf_dir, load_if_exists=False, overwrite=True, ms_before=0.25, ms_after=0.75, max_spikes_per_unit=1e6,n_jobs=1, chunk_size=30000)` 

`recording` is a 16-channel 'int16' binary recording and `sorting` is just a numpy sorting extracted from a csv file of manually sorted spike times. Here's the full traceback:

```
Traceback (most recent call last):

  File "F:\conda_parent\envs\ss096\lib\site-packages\spyder_kernels\customize\spyderpdb.py", line 889, in run
    super(SpyderPdb, self).run(cmd, globals, locals)

  File "F:\conda_parent\envs\ss096\lib\bdb.py", line 597, in run
    exec(cmd, globals, locals)

  File "f:\dropbox (weizmann institute)\ephys\code\spike sorting\objqualitymetricsman.py", line 137, in <module>
    we, metrics = get_quality_metrics(recording, sorting, wf_dir, pc_dir, metrics_file)

  File "f:\dropbox (weizmann institute)\ephys\code\spike sorting\objqualitymetricsman.py", line 88, in get_quality_metrics
    we = si.extract_waveforms(recording, sorting, wf_dir, load_if_exists=False, overwrite=True,

  File "F:\conda_parent\envs\ss096\lib\site-packages\spikeinterface\core\waveform_extractor.py", line 979, in extract_waveforms
    we.run_extract_waveforms(seed=seed, **job_kwargs)

  File "F:\conda_parent\envs\ss096\lib\site-packages\spikeinterface\core\waveform_extractor.py", line 854, in run_extract_waveforms
    wfs_arrays = extract_waveforms_to_buffers(self.recording, spikes, unit_ids, nbefore, nafter,

  File "F:\conda_parent\envs\ss096\lib\site-packages\spikeinterface\core\waveform_tools.py", line 83, in extract_waveforms_to_buffers
    distribute_waveforms_to_buffers(recording, spikes, unit_ids, wfs_arrays_info, nbefore, nafter, return_scaled,

  File "F:\conda_parent\envs\ss096\lib\site-packages\spikeinterface\core\waveform_tools.py", line 234, in distribute_waveforms_to_buffers
    processor.run()

  File "F:\conda_parent\envs\ss096\lib\site-packages\spikeinterface\core\job_tools.py", line 287, in run
    res = self.func(segment_index, frame_start, frame_stop, worker_ctx)

  File "F:\conda_parent\envs\ss096\lib\site-packages\spikeinterface\core\waveform_tools.py", line 313, in _waveform_extractor_chunk
    while (in_seg_spikes[i0]['sample_ind'] - nbefore) < 0 and (i0 != i1):

IndexError: index 62632 is out of bounds for axis 0 with size 62632
```

samuelgarcia, 2022-11-09:
HI.
I think think one spike is outside the recording lenght.
Could you check ?
We should add a better message.

Yizhar-Lab, 2022-11-14:
I checked and no spikes are outside the recording length. It seems like the error is not related to the recording length, but to the number of spikes - while there are only 62632 spikes, SI tries to index at 62632, which raises the error.

samuelgarcia, 2022-11-14:
OK. Would it be easy to get this file to try to reproduce the bug ?

alejoe91, 2022-12-28:
@Yizhar-Lab would it be possible for you to share the data triggering the error?