SpikeInterface/spikeinterface #1071: not fitted error from running export_to_phy with compute_pc_features=True

a-zmz, 2022-11-11:
Hi,

I tried to use `export_to_phy` with `compute_pc_features=True`, but I got `nofittederror`.
Please find the complete error message below.
Have anyone had this issue and how could we solve it? Thanks!

```
---------------------------------------------------------------------------                                                                                                                                                                   
_RemoteTraceback                          Traceback (most recent call last)                                                                                                                                                                   
_RemoteTraceback:                                                                                                                                                                                                                             
"""                                                                                                                                                                                                                                           
Traceback (most recent call last):                                                                                                                                                                                                            
  File "/home/amz/.conda/envs/si_tuto/lib/python3.9/concurrent/futures/process.py", line 246, in _process_worker                                                                                                                              
    r = call_item.fn(*call_item.args, **call_item.kwargs)                                                                                                                                                                                     
  File "/home/amz/.conda/envs/si_tuto/lib/python3.9/concurrent/futures/process.py", line 205, in _process_chunk                                                                                                                               
    return [fn(*args) for args in chunk]                                                                                                                                                                                                      
  File "/home/amz/.conda/envs/si_tuto/lib/python3.9/concurrent/futures/process.py", line 205, in <listcomp>                                                                                                                                   
    return [fn(*args) for args in chunk]                                                                                                                                                                                                      
  File "/home/amz/.conda/envs/si_tuto/lib/python3.9/site-packages/spikeinterface/core/job_tools.py", line 337, in function_wrapper                                                                                                            
    return _func(segment_index, start_frame, end_frame, _worker_ctx)                                                                                                                                                                          
  File "/home/amz/.conda/envs/si_tuto/lib/python3.9/site-packages/spikeinterface/postprocessing/principal_component.py", line 591, in _all_pc_extractor_chunk                                                                                 
    all_pcs[i, :, c] = pca_model[chan_ind].transform(w)                                                                                                                                                                                       
  File "/home/amz/.conda/envs/si_tuto/lib/python3.9/site-packages/sklearn/decomposition/_incremental_pca.py", line 392, in transform                                                                                                          
    return super().transform(X)                                                                                                                                                                                                               
  File "/home/amz/.conda/envs/si_tuto/lib/python3.9/site-packages/sklearn/decomposition/_base.py", line 118, in transform                                                                                                                     
    check_is_fitted(self)                                                                                                                                                                                                                     
  File "/home/amz/.conda/envs/si_tuto/lib/python3.9/site-packages/sklearn/utils/validation.py", line 1345, in check_is_fitted                                                                                                                 
    raise NotFittedError(msg % {"name": type(estimator).__name__})                                                                                                                                                                            
sklearn.exceptions.NotFittedError: This IncrementalPCA instance is not fitted yet. Call 'fit' with appropriate arguments before using this estimator.                                                                                         
"""                                                                                                                                                                                                                                           
                                                                                                                                                                                                                                              
The above exception was the direct cause of the following exception:                                                                                                                                                                          
                                                                                                                                                                                                                                              
NotFittedError                            Traceback (most recent call last)       
File ~/Documents/git/spiketutorials/Official_Tutorial_SI_0.95_Oct2022/SpikeInterface_Tutorial.py:106                                                                                                                                          
    102 print(we_curated)                                                                                                                                                                                                                     
    103 # In[5]:                                                                                                                                                                                                                              
    104                                                                                                                                                                                                                                       
    105 #we.fit()                                                                                                                                                                                                                             
--> 106 sexp.export_to_phy(we, output_folder=base_folder / 'phy_KS3',                                                                                                                                                                         
    107                    compute_amplitudes=False, copy_binary=True, remove_if_exists = True,                                                                                                                                               
    108                    **job_kwargs)                                                                                                                                                                                                      
[... cut ...]
    394         self = None                                

NotFittedError: This IncrementalPCA instance is not fitted yet. Call 'fit' with appropriate arguments before using this estimator.
> /home/amz/.conda/envs/si_tuto/lib/python3.9/concurrent/futures/_base.py(391)__get_result()                           
    389         if self._exception:
    390             try:          
--> 391                 raise self._exception
    392             finally:
    393                 # Break a reference cycle with the exception in self._exception

```

alejoe91, 2022-11-19:
Thanks for the report, @a-zmz 

Did you compute principal components before exporting to phy?

a-zmz, 2022-11-22:
Hi Alessio, I did yea, and this error only occurs if i have PC calculated before calling `export_to_phy`, somehow it does not happen if PC is calculated within `export_to_phy`.

alejoe91, 2022-11-22:
Thanks for the clarification! I think that sparsity is currently not accounted for when PC are computed for all spikes, and that's what `export_to_phy` uses. Will try to write a patch in the next few days! Sorry about that!

a-zmz, 2022-11-22:
ah i see, thanks!

closed 2022-12-12 by samuelgarcia