SpikeInterface/spikeinterface #1073: Can't load recording from json file

luiztauffer, 2022-11-14:
Tested in both `master` branch and `0.96.1` versions.

Reproducible example:
```python
import spikeinterface.extractors as se
from spikeinterface import load_extractor
from spikeinterface.core.core_tools import check_json
from pathlib import Path
import json

recording, _ = se.toy_example(
    duration=.1,
    seed=0,
    num_channels=64,
    num_segments=1
)
rec_dict = recording.to_dict()
Path('test_recording.json').write_text(json.dumps(check_json(rec_dict), indent=4), encoding='utf8')

recording = load_extractor('test_recording.json')
```

this yields:
```python
---------------------------------------------------------------------------
AssertionError                            Traceback (most recent call last)
Cell In [10], line 16
     13 rec_dict = recording.to_dict()
     14 Path('test_recording.json').write_text(json.dumps(check_json(rec_dict), indent=4), encoding='utf8')
---> 16 recording = load_extractor('test_recording.json')

File /media/luiz/storage2/Github/spikeinterface/spikeinterface/core/base.py:906, in load_extractor(file_or_folder_or_dict, base_folder)
    904     return BaseExtractor.from_dict(file_or_folder_or_dict, base_folder=base_folder)
    905 else:
--> 906     return BaseExtractor.load(file_or_folder_or_dict, base_folder=base_folder)

File /media/luiz/storage2/Github/spikeinterface/spikeinterface/core/base.py:545, in BaseExtractor.load(file_path, base_folder)
    543         print('The extractor was not dumpable')
    544         return None
--> 545     extractor = BaseExtractor.from_dict(d, base_folder=base_folder)
    546     return extractor
    548 elif file_path.is_dir():
    549     # case from a folder after a calling extractor.save(...)

File /media/luiz/storage2/Github/spikeinterface/spikeinterface/core/base.py:372, in BaseExtractor.from_dict(d, base_folder)
    370     assert base_folder is not None, 'When  relative_paths=True, need to provide base_folder'
    371     d = _make_paths_absolute(d, base_folder)
--> 372 extractor = _load_extractor_from_dict(d)
    373 folder_metadata = d.get('folder_metadata', None)
    374 if folder_metadata is not None:

File /media/luiz/storage2/Github/spikeinterface/spikeinterface/core/base.py:849, in _load_extractor_from_dict(dic)
    845     print('Versions are not the same. This might lead to errors. Use ', class_name.split('.')[0],
    846           'version', dic['version'])
    848 # instantiate extrator object
--> 849 extractor = cls(**kwargs)
    851 extractor._annotations.update(dic['annotations'])
    852 for k, v in dic['properties'].items():
    853     # print(k, v)

File /media/luiz/storage2/Github/spikeinterface/spikeinterface/core/numpyextractors.py:30, in NumpyRecording.__init__(self, traces_list, sampling_frequency, t_starts, channel_ids)
     28 def __init__(self, traces_list, sampling_frequency, t_starts=None, channel_ids=None):
     29     if isinstance(traces_list, list):
---> 30         assert all(isinstance(e, np.ndarray) for e in traces_list), 'must give a list of numpy array'
     31     else:
     32         assert isinstance(traces_list, np.ndarray), 'must give a list of numpy array'

AssertionError: must give a list of numpy array
```

samuelgarcia, 2022-11-14:
Hi Luiz.
toy example generate Numpy based recording/sorting.
They are not dumpable because they use in memory buffer.
So the to_dict() is working but making this persistent in a json file is not possible.

If you want to have persistent toy you need to have 
```python
rec = rec.save(...)
sorting = sorting.save(...)
```

dump_to_json() should raise an error.

luiztauffer, 2022-11-14:
hi @samuelgarcia 
but a json file is being produced with the data, but the data is being stored as lists instead of np.arrays... it should be possible to read back from it, right?
`rec.save()` doesn't seem to give me an option to save is as JSON directly

This is making the basic example of running sorters from a container fail, since the container code presumes the recordings are stored as json files

samuelgarcia, 2022-11-14:
The `to_dict()` / `to_json()` is the concept of "dumpability". In other words, saving the kwargs to construct back the extractor with an init. It is not something for "saving". Saving is done with `save()` because it also also copy the data buffer.

The to_dict in the case of NumpyRecording contain the buffer as an array because it can be seriliaze for parralell processing but not for using this as a format. It is a corner case.

The `Path('test_recording.json').write_text(json.dumps(check_json(rec_dict), indent=4), encoding='utf8')` should be done with `rec.dump_to_json()` which is the [same](https://github.com/SpikeInterface/spikeinterface/blob/master/spikeinterface/core/base.py#L470) but with some more checking (that is not done here)

I gues that `to_dict()` and then saving with pickle should work but I would absolutly **not** recommand it.

alejoe91, 2022-12-28:
Behavior of handling non-dumpable object has been improved here: #1131

closed 2022-12-28 by alejoe91