SpikeInterface/spikeinterface #1077: Can't run sorters using Docker

roybens, 2022-11-17:
I tried both mountainsort and kilosort3 and i get the same error. It is probably a simple one since it is occuring during cleanup:

sorting_KS3 = ss.run_kilosort3(recording, docker_image="spikeinterface/kilosort3-compiled-base")

FileNotFoundError                         Traceback (most recent call last)
Cell In [5], line 1
----> 1 sorting_MS4 = ss.run_mountainsort4(recording, docker_image="spikeinterface/mountainsort4-base:latest")

File ~/.local/lib/python3.8/site-packages/spikeinterface/sorters/runsorter.py:656, in run_mountainsort4(*args, **kwargs)
    655 def run_mountainsort4(*args, **kwargs):
--> 656     return run_sorter('mountainsort4', *args, **kwargs)

File ~/.local/lib/python3.8/site-packages/spikeinterface/sorters/runsorter.py:137, in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, singularity_image, with_output, **sorter_params)
    135         else:
    136             container_image = singularity_image
--> 137     return run_sorter_container(
    138         container_image=container_image,
    139         mode=mode,
    140         **common_kwargs,
    141     )
    143 return run_sorter_local(**common_kwargs)

File ~/.local/lib/python3.8/site-packages/spikeinterface/sorters/runsorter.py:559, in run_sorter_container(sorter_name, recording, mode, container_image, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, extra_requirements, **sorter_params)
    556 container_client.stop()
    558 # clean useless files
--> 559 os.remove(parent_folder / 'in_container_recording.json')
    560 os.remove(parent_folder / 'in_container_params.json')
    561 os.remove(parent_folder / 'in_container_sorter_script.py')

FileNotFoundError: [Errno 2] No such file or directory: '/home/mxwbio/Desktop/MEA_Analysis/in_container_recording.json'
---------------------------------------------------------------------------
FileNotFoundError                         Traceback (most recent call last)
Cell In [6], line 1
----> 1 sorting_KS3 = ss.run_kilosort3(recording, docker_image="spikeinterface/kilosort3-compiled-base")

File ~/.local/lib/python3.8/site-packages/spikeinterface/sorters/runsorter.py:691, in run_kilosort3(*args, **kwargs)
    690 def run_kilosort3(*args, **kwargs):
--> 691     return run_sorter('kilosort3', *args, **kwargs)

File ~/.local/lib/python3.8/site-packages/spikeinterface/sorters/runsorter.py:137, in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, singularity_image, with_output, **sorter_params)
    135         else:
    136             container_image = singularity_image
--> 137     return run_sorter_container(
    138         container_image=container_image,
    139         mode=mode,
    140         **common_kwargs,
    141     )
    143 return run_sorter_local(**common_kwargs)

File ~/.local/lib/python3.8/site-packages/spikeinterface/sorters/runsorter.py:559, in run_sorter_container(sorter_name, recording, mode, container_image, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, extra_requirements, **sorter_params)
    556 container_client.stop()
    558 # clean useless files
--> 559 os.remove(parent_folder / 'in_container_recording.json')
    560 os.remove(parent_folder / 'in_container_params.json')
    561 os.remove(parent_folder / 'in_container_sorter_script.py')

FileNotFoundError: [Errno 2] No such file or directory: '/home/mxwbio/Desktop/MEA_Analysis/in_container_recording.json'

alejoe91, 2022-11-19:
Hi @roybens 

Can you run with `verbose=True` and report the output? For Linux, you might be getting a permission error if you didn't allowed docker to run without prove ledges..

Does this work? `docker run hello-world`?

roybens, 2022-11-20:
Yes that worked, Thank you!

closed 2022-11-20 by roybens